bluesky/ophyd-async #13: pvget .put doesn't work with enums

RAYemelyanova, 2023-09-12:
Working on a panda on P38, I've come across as scenario where a .set on a SignalRW works with channel access, but doesn't work with pvaccess.

That is,

```python
In[3]: source = "BL38P-PANDA:SEQ1:PRESCALE:UNITS"

In[4]: pv_signal = epics_signal_rw(None, "pva://" + source)
In[5]: ca_signal = epics_signal_rw(None, source)

In[6]: await ca_signal.set("s")

In[7]: await ca_signal.get_value()
Out[7]: <GeneratedChoices.s: 's'>

In[8]: await ca_signal.set("ms")

In[9]: await ca_signal.get_value()
Out[9]: <GeneratedChoices.ms: 'ms'>

In[10]: await pv_signal.set("s")
Out[9]: <GeneratedChoices.ms: 'ms'>
```

coretl, 2023-09-13:
Reported upstream as https://github.com/mdavidsaver/p4p/issues/123

coretl, 2024-04-16:
Fixed upstream

closed 2024-04-16 by coretl