bluesky/ophyd-async #23: flaky tests

rosesyrett, 2023-10-04:
Sometimes CI tests fail. Flaky tests are bad and we should aim to not have any. Here is an example of such a test failing on one instance, but passing if it's rerun:
https://github.com/bluesky/ophyd-async/actions/runs/6405250037/job/17387404995?pr=22

It seems the offending error is this:

```
Traceback (most recent call last):
157
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/site-packages/_pytest/runner.py", line 341, in from_call
158
    result: Optional[TResult] = func()
159
                                ^^^^^^
160
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/site-packages/_pytest/runner.py", line 262, in <lambda>
161
    lambda: ihook(item=item, **kwds), when=when, reraise=reraise
162
            ^^^^^^^^^^^^^^^^^^^^^^^^
163
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/site-packages/pluggy/_hooks.py", line 493, in __call__
164
    return self._hookexec(self.name, self._hookimpls, kwargs, firstresult)
165
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
166
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/site-packages/pluggy/_manager.py", line 115, in _hookexec
167
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
168
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
169
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/site-packages/pluggy/_callers.py", line 130, in _multicall
170
    teardown[0].send(outcome)
171
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/site-packages/_pytest/unraisableexception.py", line 88, in pytest_runtest_call
172
    yield from unraisable_exception_runtest_hook()
173
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/site-packages/_pytest/unraisableexception.py", line 78, in unraisable_exception_runtest_hook
174
    warnings.warn(pytest.PytestUnraisableExceptionWarning(msg))
175
pytest.PytestUnraisableExceptionWarning: Exception ignored in: <coroutine object SignalR.unstage at 0x7ff6e4e03920>
176

177
Traceback (most recent call last):
178
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/warnings.py", line 537, in _warn_unawaited_coroutine
179
    warn(msg, category=RuntimeWarning, stacklevel=2, source=coro)
180
RuntimeWarning: coroutine 'SignalR.unstage' was never awaited
181
Coroutine created at (most recent call last)
182
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/threading.py", line 995, in _bootstrap
183
    self._bootstrap_inner()
184
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
185
    self.run()
186
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/threading.py", line 975, in run
187
    self._target(*self._args, **self._kwargs)
188
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/base_events.py", line 607, in run_forever
189
    self._run_once()
190
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/base_events.py", line 1914, in _run_once
191
    handle._run()
192
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/events.py", line 80, in _run
193
    self._context.run(self._callback, *self._args)
194
  File "/home/runner/work/ophyd-async/ophyd-async/src/ophyd_async/core/_device/standard_readable.py", line 51, in unstage
195
    await sig.unstage().task
196
  File "/home/runner/work/ophyd-async/ophyd-async/src/ophyd_async/core/async_status.py", line 80, in wrap_f
197
    return AsyncStatus(f(self))
```

with the following logs recorded:

```
11:19:29,664 ERROR (MainThread) Task was destroyed but it is pending!
source_traceback: Object created at (most recent call last):
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/threading.py", line 995, in _bootstrap
    self._bootstrap_inner()
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/base_events.py", line 607, in run_forever
    self._run_once()
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/base_events.py", line 1914, in _run_once
    handle._run()
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/site-packages/bluesky/run_engine.py", line 1588, in _run
    new_response = await coro(msg)
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/site-packages/bluesky/run_engine.py", line 2420, in _unstage
    ret = obj.unstage()
  File "/home/runner/work/ophyd-async/ophyd-async/src/ophyd_async/core/async_status.py", line 80, in wrap_f
    return AsyncStatus(f(self))
  File "/home/runner/work/ophyd-async/ophyd-async/src/ophyd_async/core/async_status.py", line 23, in __init__
    self.task = asyncio.create_task(awaitable)  # type: ignore
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/tasks.py", line 374, in create_task
    task = loop.create_task(coro)
task: <Task pending name='Task-604' coro=<HDFStreamerDet.unstage() running at /home/runner/work/ophyd-async/ophyd-async/src/ophyd_async/epics/areadetector/hdf_streamer_det.py:167> wait_for=<Task finished name='Task-608' coro=<StandardReadable.unstage() done, defined at /home/runner/work/ophyd-async/ophyd-async/src/ophyd_async/core/_device/standard_readable.py:48> result=None created at /opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/tasks.py:374> cb=[AsyncStatus._run_callbacks()] created at /opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/tasks.py:374>
```

callumforrester, 2024-04-16:
I haven't seen flaky tests for several months so going to close this. If they come back, I've wondered about detecting them with infrequent use of `pytest-repeat`

DiamondJoseph, 2024-09-10:
Flaky tests have re-appeared, I'm going to re-open this ticket.

DiamondJoseph, 2024-09-11:
pytest plugin for generating CTRF report for the above https://pypi.org/project/pytest-json-ctrf/

callumforrester, 2024-09-11:
I think all runs, why not?

DiamondJoseph, 2024-09-11:
Identifying flaky tests vs. tests that were temporarily broken by unfinished PRs?

DiamondJoseph, 2024-09-11:
@callumforrester I've enabled on this draft PR (scroll down to summary section): it hasn't marked any tests as flaky yet, and doesn't provide any easier method of seeing what's flaky/broken.

@coretl for comment too

https://github.com/bluesky/ophyd-async/actions/runs/10809455998?pr=567

coretl, 2024-10-14:
Tango tests are apparently flaky because of Tango test infrastructure and there is nothing we can do to fix that. I suggest we report test failures on this ticket and make PRs that `@pytest.mark.flaky(reruns=2)` individual flaky tests until the CI is reliable again.

The first PR that does this needs to add `pytest-rerunfailures` to the dev dependencies to get the marker.

coretl, 2025-04-04:
Flaky test on windows 3.10:
```
________________________ test_observe_value_times_out _________________________
Traceback (most recent call last):
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\runner.py", line 341, in from_call
    result: TResult | None = func()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\runner.py", line 242, in <lambda>
    lambda: runtest_hook(item=item, **kwds), when=when, reraise=reraise
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_hooks.py", line 513, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_manager.py", line 120, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 182, in _multicall
    return outcome.get_result()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_result.py", line 100, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 167, in _multicall
    teardown.throw(outcome._exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\threadexception.py", line 92, in pytest_runtest_call
    yield from thread_exception_runtest_hook()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\threadexception.py", line 68, in thread_exception_runtest_hook
    yield
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 167, in _multicall
    teardown.throw(outcome._exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\unraisableexception.py", line 95, in pytest_runtest_call
    yield from unraisable_exception_runtest_hook()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\unraisableexception.py", line 70, in unraisable_exception_runtest_hook
    yield
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 167, in _multicall
    teardown.throw(outcome._exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\logging.py", line 846, in pytest_runtest_call
    yield from self._runtest_for(item, "call")
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\logging.py", line 829, in _runtest_for
    yield
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 167, in _multicall
    teardown.throw(outcome._exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\capture.py", line 898, in pytest_runtest_call
    return (yield)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 167, in _multicall
    teardown.throw(outcome._exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\skipping.py", line 257, in pytest_runtest_call
    return (yield)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 103, in _multicall
    res = hook_impl.function(*args)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\runner.py", line 174, in pytest_runtest_call
    item.runtest()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pytest_asyncio\plugin.py", line 549, in runtest
    super().runtest()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\python.py", line 1627, in runtest
    self.ihook.pytest_pyfunc_call(pyfuncitem=self)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_hooks.py", line 513, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_manager.py", line 120, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 182, in _multicall
    return outcome.get_result()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_result.py", line 100, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 103, in _multicall
    res = hook_impl.function(*args)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\python.py", line 159, in pytest_pyfunc_call
    result = testfunction(**testargs)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pytest_asyncio\plugin.py", line 1069, in inner
    _loop.run_until_complete(task)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\asyncio\base_events.py", line 649, in run_until_complete
    return future.result()
  File "D:\a\ophyd-async\ophyd-async\tests\core\test_observe.py", line 72, in test_observe_value_times_out
    assert time.time() - start == pytest.approx(0.2, abs=0.05)
AssertionError: assert 0.32814812660217285 == 0.2 � 5.0e-02
  
  comparison failed
  Obtained: 0.32814812660217285
  Expected: 0.2 � 5.0e-02
------------------------------ Captured log call ------------------------------
10:18:49,770 WARNING (MainThread) Executing <Task pending name='Task-1658' coro=<test_observe_value_times_out.<locals>.tick() running at D:\a\ophyd-async\ophyd-async\tests\core\test_observe.py:57> wait_for=<Future pending cb=[Task.task_wakeup()] created at C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\asyncio\base_events.py:429> created at C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\asyncio\tasks.py:337> took 0.125 seconds
```

coretl, 2025-08-01:
Flaky test for ubuntu py3.12
```
_____________________ test_move_profiles[-0.025-expected3] _____________________
Traceback (most recent call last):
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/runner.py", line 344, in from_call
    result: TResult | None = func()
                             ^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/runner.py", line 246, in <lambda>
    lambda: runtest_hook(item=item, **kwds), when=when, reraise=reraise
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_hooks.py", line 512, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_manager.py", line 120, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 167, in _multicall
    raise exception
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/logging.py", line 850, in pytest_runtest_call
    yield
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 53, in run_old_style_hookwrapper
    return result.get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_result.py", line 103, in get_result
    raise exc.with_traceback(tb)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 38, in run_old_style_hookwrapper
    res = yield
          ^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/capture.py", line 900, in pytest_runtest_call
    return (yield)
            ^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 53, in run_old_style_hookwrapper
    return result.get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_result.py", line 103, in get_result
    raise exc.with_traceback(tb)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 38, in run_old_style_hookwrapper
    res = yield
          ^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/skipping.py", line 263, in pytest_runtest_call
    return (yield)
            ^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 121, in _multicall
    res = hook_impl.function(*args)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/runner.py", line 178, in pytest_runtest_call
    item.runtest()
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pytest_asyncio/plugin.py", line 426, in runtest
    super().runtest()
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/python.py", line 1671, in runtest
    self.ihook.pytest_pyfunc_call(pyfuncitem=self)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_hooks.py", line 512, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_manager.py", line 120, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 167, in _multicall
    raise exception
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
[... 9 lines cut ...]
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 121, in _multicall
    res = hook_impl.function(*args)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/python.py", line 157, in pytest_pyfunc_call
    result = testfunction(**testargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...] phyd_async/core/_status.py:148> wait_for=<Task pending name='Task-15129' coro=<SignalW.set() running at /opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/ophyd_async/core/_signal.py:277> cb=[AsyncStatusBase._run_callbacks(), Task.task_wakeup()] created at /opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/asyncio/tasks.py:420> cb=[AsyncStatusBase._run_callbacks()] created at /opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/asyncio/tasks.py:420> took 0.118 seconds
```

GDYendell, 2025-08-04:
Flaky test on ubuntu 3.12
```
=================================== FAILURES ===================================
_____________________ test_move_profiles[-0.025-expected3] _____________________
Traceback (most recent call last):
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/runner.py", line 344, in from_call
    result: TResult | None = func()
                             ^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/runner.py", line 246, in <lambda>
    lambda: runtest_hook(item=item, **kwds), when=when, reraise=reraise
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_hooks.py", line 512, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_manager.py", line 120, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 167, in _multicall
    raise exception
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/logging.py", line 850, in pytest_runtest_call
    yield
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 53, in run_old_style_hookwrapper
    return result.get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_result.py", line 103, in get_result
    raise exc.with_traceback(tb)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 38, in run_old_style_hookwrapper
    res = yield
          ^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/capture.py", line 900, in pytest_runtest_call
    return (yield)
            ^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 53, in run_old_style_hookwrapper
    return result.get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_result.py", line 103, in get_result
    raise exc.with_traceback(tb)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 38, in run_old_style_hookwrapper
    res = yield
          ^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/skipping.py", line 263, in pytest_runtest_call
    return (yield)
            ^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 121, in _multicall
    res = hook_impl.function(*args)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/runner.py", line 178, in pytest_runtest_call
    item.runtest()
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pytest_asyncio/plugin.py", line 426, in runtest
    super().runtest()
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/_pytest/python.py", line 1671, in runtest
    self.ihook.pytest_pyfunc_call(pyfuncitem=self)
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_hooks.py", line 512, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_manager.py", line 120, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.12.11/x64/lib/python3.12/site-packages/pluggy/_callers.py", line 167, in _multicall

```

GDYendell, 2025-08-04:
Flaky test windows 3.10
```
================================== FAILURES ===================================
____________ test_many_individual_device_connects_not_slow[False] _____________
Traceback (most recent call last):
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\runner.py", line 344, in from_call
    result: TResult | None = func()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\runner.py", line 246, in <lambda>
    lambda: runtest_hook(item=item, **kwds), when=when, reraise=reraise
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_hooks.py", line 512, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_manager.py", line 120, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 167, in _multicall
    raise exception
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\logging.py", line 850, in pytest_runtest_call
    yield
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 53, in run_old_style_hookwrapper
    return result.get_result()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_result.py", line 103, in get_result
    raise exc.with_traceback(tb)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 38, in run_old_style_hookwrapper
    res = yield
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\capture.py", line 900, in pytest_runtest_call
    return (yield)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 53, in run_old_style_hookwrapper
    return result.get_result()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_result.py", line 103, in get_result
    raise exc.with_traceback(tb)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 38, in run_old_style_hookwrapper
    res = yield
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\skipping.py", line 263, in pytest_runtest_call
    return (yield)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 121, in _multicall
    res = hook_impl.function(*args)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\runner.py", line 178, in pytest_runtest_call
    item.runtest()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pytest_asyncio\plugin.py", line 426, in runtest
    super().runtest()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\python.py", line 1671, in runtest
    self.ihook.pytest_pyfunc_call(pyfuncitem=self)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_hooks.py", line 512, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_manager.py", line 120, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 167, in _multicall
    raise exception
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 139, in _multicall
    teardown.throw(exception)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 53, in run_old_style_hookwrapper
    return result.get_result()
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_result.py", line 103, in get_result
    raise exc.with_traceback(tb)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 38, in run_old_style_hookwrapper
    res = yield
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pluggy\_callers.py", line 121, in _multicall
    res = hook_impl.function(*args)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\_pytest\python.py", line 157, in pytest_pyfunc_call
    result = testfunction(**testargs)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\site-packages\pytest_asyncio\plugin.py", line 642, in inner
    _loop.run_until_complete(task)
  File "C:\hostedtoolcache\windows\Python\3.10.11\x64\lib\asyncio\base_events.py", line 649, in run_until_complete
    return future.result()
  File "D:\a\ophyd-async\ophyd-async\tests\core\test_device.py", line 220, in test_many_individual_device_connects_not_slow
    assert duration < 1
AssertionError: assert 1.0779999999999745 < 1
``