bluesky/ophyd-async #40: Soft signals vs attributes

coretl, 2023-10-26:
In the `TetrAMM` we have attribute setting:
https://github.com/DiamondLightSource/dodal/blob/5ff8127eb378c1087c66cdb7c95eb5d2424c9456/src/dodal/devices/tetramm.py#L113-L120

This is then set in a plan, which means you can't listify it without having side effects. We could turn this into a soft signal and do `bps.abs_set` on it instead, but is it worth it? Or do we never intend to preprocess or listify plans...

If we do want a soft signal we should make it easier to construct one and write some docs around why you would use one.

@callumforrester @rosesyrett @tpoliaw thoughts?

callumforrester, 2023-10-27:
There has already been some debate about this so input welcome. 

The tradeoff is obviously the potential for validation vs the potential restriction from a plan having to be listifyable. One argument is to take the approach that the Python language itself has taken with type hints: provide a listifying, validated route but make it optional if people still want to go crazy. It then comes down to the individual developer and how much they care about static verification. 

I'm not hugely in favour of this as a defender of type systems, and static verification systems in general. However I suspect it satisfies the largest number of use cases. It also provides a clear migration path from side-effectful to side-effecless plans, because you start with a hacky plan with side effects and, if they are useful, move them into the `RunEngine` over time.

coretl, 2023-10-31:
I guess that we can still use pre-processors like the DirectoryProvider one, as long as we are aware that any preprocessor we apply may change the yielding of the messages with respect to the attribute setting...

tacaswell, 2023-11-06:
It is the design goal to push all state changes through the RE to make it safe to simulate plans without affecting reality, but given that this is Python there is no way to actually enforce that (I mean you can do `subprocess.run(['rm', '-rf', '*'])` ....) .

In ophyd-sync we have 
https://github.com/bluesky/ophyd/blob/9a0bf3f5fd8c98ee23e5d51c35db695c86a6a921/ophyd/signal.py#L2192-L2271 for bridging the attribute / signal gap which can preserve both worlds (at the cost of two names).  If you do not strictly need the dotted name access I would go with a soft-signal as then the path to externalizing that state is gentle.

callumforrester, 2024-04-16:
@coretl I think this is resolved to my satisfaction but feel free to reopen

closed 2024-04-16 by callumforrester