bluesky/ophyd-async #44: Signal Kind

whs92, 2023-10-31:
In ophyd there is the concept of attribute `kind` https://nsls-ii.github.io/ophyd/signals.html?highlight=kind#kind 

This is quite useful when looking at data, separating signals into stuff that's really important and other stuff that is just useful for context. Differentiating `normal` and `hinted` signals is also useful for plotting

Is this also planned for ophyd-async, or is there another method for differentiating signals?

coretl, 2023-11-01:
When designing ophyd-async, we tried to separate the interface to the underlying Devices from the logic that would use them as much as possible. I class "what to put in `read()` and `read_configuration()`" and "what to put in `hints`" as logic rather than interface, so didn't attach anything like `kind` to signals. 

To answer "what to put in `read()` and `read_configuration()`", I made `StandardReadable` with:
https://github.com/bluesky/ophyd-async/blob/f1b4aba227ee6b568718c170ced6378a744abf99/src/ophyd_async/core/standard_readable.py#L23-L38

which is used like this:
https://github.com/bluesky/ophyd-async/blob/f1b4aba227ee6b568718c170ced6378a744abf99/src/ophyd_async/epics/demo/__init__.py#L31-L43

I didn't however provide any helpers in `StandardReadable` to help with `hints`. You can always make the `hints` property yourself:
```python
class Sensor(StandardReadable):
    ...
    @property
    def hints(self):
        return {"fields": [self.value.name]}
```

However, it occurs to me that we could add this to `StandardReadable.set_readable_signals` too, although the set of `uncached` and `hinted` may overlap. Maybe we need to change the signature to:
```python
     def set_readable_signals( 
         self, 
         config: Sequence[SignalR] = (), 
         read: Sequence[SignalR] | Dict[SignalR, SignalReadOptions] = (),   # Sequence means use default values of SignalReadOptions
     ): 
```
where `SignalReadOptions` might be:
```python
@dataclass
class SignalReadOptions:
    cached: True
    hinted: False

# For convenience
HINTED = SignalReadOptions(hinted=True)
UNCACHED = SignalReadOptions(cached=False)
UNCACHED_HINTED = SignalReadOptions(cached=False, hinted=True)
```

So the example would become:
```python
class Sensor(StandardReadable):
    def __init__(self, prefix: str, name="") -> None: 
        self.value = epics_signal_r(float, prefix + "Value") 
        self.mode = epics_signal_rw(EnergyMode, prefix + "Mode") 
        self.set_readable_signals( 
            config=[self.mode], 
            read={self.value: HINTED}, 
        ) 
```

What do you reckon?

whs92, 2023-11-02:
I like this solution you propose:

```python
class Sensor(StandardReadable):
    def __init__(self, prefix: str, name="") -> None: 
        self.value = epics_signal_r(float, prefix + "Value") 
        self.mode = epics_signal_rw(EnergyMode, prefix + "Mode") 
        self.set_readable_signals( 
            config=[self.mode], 
            read={self.value: HINTED}, 
        ) 
```

coretl, 2023-11-03:
@danielballan @tacaswell any thoughts on this?

coretl, 2024-04-30:
Completed in https://github.com/bluesky/ophyd-async/issues/223

closed 2024-04-30 by coretl