bluesky/ophyd-async #49: Add function for callback to be removed from sim signal

DominicOram, 2023-11-01:
When writing tests I want to simulate some basic signal behaviour e.g. writing to one signal will put to another. However, I would like to make sure this behaviour is properly cleaned up between tests. I can simulate with `set_sim_callback` and `set_sim_value` but there is no way to remove this callback.

## Acceptance Criteria
* There is a function to remove callbacks from simulated signals

coretl, 2023-11-02:
I like this being a context, then making it fire on put rather than on every update, as that is what the tests want

DominicOram, 2023-11-02:
Taken myself off this as @coretl said they will find someone else to pick it up. It's not urgent from our perspective

coretl, 2023-12-12:
I think this will be fixed if we have a generic Mock backend in #94

coretl, 2024-06-17:
Fixed by #94

closed 2024-06-17 by coretl