bluesky/ophyd-async #52: Discussion: Should children automaticaaly traverse more of the object

DominicOram, 2023-11-01:
Currently the `children` function on a device checks if the `Device` has any direct attributes that are `Devices`. This results in the following:
```python
class MyDevice(Device):
    def __init__():
        self.channels = [epics_signal_r(float, f"A"), epics_signal_r(float, f"B")]

my_device.connect() # channels will not get connected here
```
The fix for this is something like:
```python
class MyDevice(Device):
    def __init__():
        self.channels = [epics_signal_r(float, f"A"), epics_signal_r(float, f"B")]
   def children():
       for signal in self.channels:
           yield signal
       return super().children()
```

@coretl, @callumforrester - I feel like we may have discussed this already but needs writing down

## Acceptance Criteria
* We decide if `children` should be more clever
* Decision and reasoning is documented
* If the above behaviour remains it should be documented as a gotcha

coretl, 2023-11-02:
If you use a `DeviceVector` (and please write docs for it too!) then this will work

callumforrester, 2024-04-16:
Docs for `DeviceVector` in #183 

@DominicOram does it satisfy your requirements?

DominicOram, 2024-04-16:
Yes, happy to close this off

closed 2024-04-16 by DominicOram