bluesky/ophyd-async #64: Creating too many statuses

callumforrester, 2023-11-10:
In reference to https://github.com/DiamondLightSource/blueapi/pull/331

It appears that a very large number of status objects are sometimes creating during long-running flyscans. Can reproduce by running the linkam plan: https://github.com/DiamondLightSource/i22-bluesky/blob/move_ophyd_async/src/i22_bluesky/plans/linkam.py and subscribing to the RE waiting hook.

At this point this is a very mysterious bug and this issue is for further investigation.

coretl, 2023-11-14:
Discovered that this is caused by repeatedly:

- Calling `bps.wait(complete_group, timeout=0.5)`
- blueapi notified of `complete_status` and adds watcher
- wait times out
- blueapi notified `None` and does nothing about it

Each time round the loop, blueapi added a watcher to the same status, and ophyd-async didn't check for duplicates.

Now we need to think about what is the correct behaviour...

coretl, 2023-12-12:
Made a new issue to fix this in https://github.com/bluesky/bluesky/issues/1641

closed 2023-12-12 by coretl