bluesky/ophyd-async #72: Flyer: no existing protection against grouping of stream resources/datums across points

rosesyrett, 2023-11-13:
Problem
======
At the moment, in the flyer collect_asset_docs method, the flyer collects together stream datums, and emits them after all stream resources have been emitted:

```python
        current_frame = self._current_frame
        stream_datums: List[Asset] = []
        async for asset in self._detector_group_logic.collect_asset_docs():
            name, doc = asset
            if name == "stream_datum":
                current_frame = doc["indices"]["stop"] + self._offset
                # Defer stream_datums until all stream_resources have been produced
                # In a single collect, all the stream_resources are produced first
                # followed by their stream_datums
                stream_datums.append(asset)
            else:
                yield asset
        for asset in stream_datums:
            yield asset
```

The detector group logic has a method, collect_asset_docs, which just spits out all the stream datums/resources for each writer involved in the fly scan (i.e .each hdf writer): 

```python
        for writer in self._writers:
            async for doc in writer.collect_stream_docs(indices_written):
                yield doc
```

In a flyscan, stream resources and datums are grouped for each collection point in the flyscan. i.e. with each set of triggers sent to the detectors. It is *technically possible* that, if we have lots of writers involved in the fly scan, we could still be spitting out stream datums/resources for one collection point as the trigger for the next one has been set. This could lead to unexpected results, e.g. if we expect n stream datums/stream resources per collection point we could see the following set of documents being produced:

start
descriptor
stream resource (2n)
stream datum (2n)

Instead of, what we would really want:

start
descriptor
stream resource (n)
stream datum (n)
stream resource (n)
stream datum (n)

Possible fix
=========
We could fix this by making the writers emit stream datums/resources in parallel, instead of in series as is currently happening. Ultimately though, we could be asking our flyer (i.e. the panda sending triggers, in i22's case) to be sending out triggers faster than the python logic to produce stream resources/datums can run. At which point we run into a bigger architectural problem. However, as it stands, if we have a huge list of detectors and set up the panda to send lots of triggers in quick succession, we can reach this problem. I don't know if it is worth including a warning just by looking at how quickly those triggers have been set up.

Therefore, it is *technically possible* that, when one writer

coretl, 2023-11-13:
Just to clarify, fast triggers is not a problem, we include `indices_written` so that only one `stream_datum` is written of the correct size, even if the detector has moved on. The only time this would be a problem is if a flush happened that encompassed the writing of 2 `stream_resources`, i.e. in 0.5s we are creating 2 different HDF files from the same detector. We have no use-case for this, so I think we should either catch this case and error, or do nothing and let downstream deal with it if it ever did occur

coretl, 2024-02-28:
This is no longer a problem as we have split the detectors off from the flyers

closed 2024-02-28 by coretl