bluesky/ophyd-async #73: Don't harcode HDFFile dataset location

rosesyrett, 2023-11-13:
At the moment there is just a hardcoded string in the HDFWriter indicating any data written by scans will be stored in 'entry/data/data'. 
This is fine for our current use case (i22), except it'd be better to not hardcode this and consider the wider implications of that, i.e. do we make this a soft signal and have that set in the plan?

coretl, 2023-11-13:
ADCore HDF writer will always write there unless you override it by providing a NeXus template xml for it to use. We should ensure that is blank and then we can rely on it always being `/entry/data/data`

coretl, 2024-04-16:
Still need to ensure that we add `self.xml_file_name = ad_rw(prefix + "XMLFileName")` and set it to `""` before we open the HDF file as per https://areadetector.github.io/areaDetector/ADCore/NDFileHDF5.html#parameters-and-records

Villtord, 2024-04-17:
How can I test if I change path in HDFWriter? I would like to investigate "real" (in terms of existence) nexus or hdf file structure before/after my change. Will any of the detectors from docs/user/examples/epics_demo.py write actual hdf file in a dev container?