bluesky/ophyd-async #81: should we be able to connect devices without the run engine?

rosesyrett, 2023-11-14:
Part of the appeal of ophyd v1 is that you can use it stand-alone, without the run engine or bluesky. This is not true of ophyd-async, only because we need to bluesky event loop to be running to connect devices. Could we not change the connection logic so that;

1. You can connect your devices in a separate (i.e. local) event loop,
2. When you make the run engine, in the background something works out that your running event loop should be used for the run engine instead.

coretl, 2023-11-14:
At the moment the RE is only require if you do `with DeviceCollector():` and if you do `motor.move()` rather than `await motor.set()`. For the first, if you do `async with DeviceCollector():` within an event loop it will work fine, and for the second I think we should ditch `motor.move()` and do this with magics instead

coretl, 2023-11-14:
And you can pass `loop` to RE and it will use that one, so maybe it should default to using the current one rather than making its own if it is within an event loop?

callumforrester, 2024-01-29:
+1 to defaulting to current loop, would be good minimize coupling between ophyd async and bluesky

evalott100, 2024-02-15:
@coretl, @callumforrester, @OCopping and I had a meeting on this. we agreed to leave things how they are, but add documentation. A summary of how things currently work:

[image]

## `with DeviceCollector`

On exit we call in the bluesky `event-loop`

``` python
    def __exit__(self, type_, value, traceback):
        self._objects_on_exit = self._caller_locals()
        return call_in_bluesky_event_loop(self._on_exit())
```

`call_in_bluesky_event_loop` will fail if the bluesky event loop isn't running.

## `async with DeviceCollector`

We don't force the bluesky event-loop, it's down to the user to choose to run in the same event loop.
 
``` python
    async def _on_exit(self) -> None:
        # Name and kick off connect for devices
        connect_coroutines: Dict[str, Coroutine] = {}
        for name, obj in self._objects_on_exit.items():
            if name not in self._names_on_enter and isinstance(obj, Device):
                if self._set_name and not obj.name:
                    obj.set_name(name)
                if self._connect:
                    connect_coroutines[name] = obj.connect(
                        self._sim, timeout=self._timeout
                    )

        # Connect to all the devices
        if connect_coroutines:
            await wait_for_connection(**connect_coroutines)

    async def __aexit__(self, type, value, traceback):
        self._objects_on_exit = self._caller_locals()
        await self._on_exit()
```

PR #114 (2024-02-16): Add user information on the `DeviceCollector` event-loop
Closes #81 #112 

* Adds documentation on the event-loop used in `DeviceCollector`
* Adds better error messages
closed 2024-02-16 by evalott100