bluesky/ophyd-async #106: Issue trying to run linkam_plan for i22: prepare

rosesyrett, 2024-01-23:
Just tried to run the `linkam_plan` on i22, and blueapi had the following error trace:

```
ERROR:blueapi.worker.reworker:<AsyncStatus errored>
Traceback (most recent call last):
  File "/dls_sw/i22/software/blueapi/scratch/ophyd-async/src/ophyd_async/core/flyer.py", line 224, in _prepare
    await asyncio.gather(
  File "/dls_sw/i22/software/blueapi/scratch/ophyd-async/src/ophyd_async/core/flyer.py", line 112, in ensure_armed
    self._arm_statuses = await gather_list(
                         ^^^^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/ophyd-async/src/ophyd_async/core/utils.py", line 104, in gather_list
    return await asyncio.gather(*coros)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/dodal/src/dodal/devices/tetramm.py", line 131, in arm
    raise Exception("Tetramm has no concept of setting a number of exposures.")
Exception: Tetramm has no concept of setting a number of exposures.
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/venv/lib/python3.11/site-packages/blueapi/worker/reworker.py", line 235, in _cycle
    self._current.task.do_task(self._ctx)
  File "/venv/lib/python3.11/site-packages/blueapi/worker/task.py", line 33, in do_task
    ctx.run_engine(wrapped_plan_generator)
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/run_engine.py", line 904, in __call__
    plan_return = self._resume_task(init_func=_build_task)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/run_engine.py", line 1043, in _resume_task
    raise exc
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/run_engine.py", line 1673, in _run
    raise err
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/run_engine.py", line 1533, in _run
    msg = self._plan_stack[-1].send(resp)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/venv/lib/python3.11/site-packages/blueapi/core/context.py", line 76, in wrap
    yield from wrapped_plan
  File "/venv/lib/python3.11/site-packages/blueapi/preprocessors/attach_metadata.py", line 36, in attach_metadata
    yield from bpp.inject_md_wrapper(
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/preprocessors.py", line 752, in inject_md_wrapper
    return (yield from msg_mutator(plan, _inject_md))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/preprocessors.py", line 263, in msg_mutator
    msg = plan.send(_s)
          ^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/i22-bluesky/src/i22_bluesky/plans/linkam.py", line 159, in linkam_plan
    rs_uid = yield from inner_linkam_plan()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/preprocessors.py", line 675, in dec_inner
    ret = yield from plan
          ^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/preprocessors.py", line 675, in dec_inner
    ret = yield from plan
          ^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/utils/__init__.py", line 1203, in dec_inner
    return (yield from plan)
            ^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/preprocessors.py", line 985, in stage_wrapper
    return (yield from finalize_wrapper(inner(), unstage_devices()))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/preprocessors.py", line 535, in finalize_wrapper
    ret = yield from plan
          ^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/preprocessors.py", line 983, in inner
    return (yield from plan)
            ^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/utils/__init__.py", line 1203, in dec_inner
    return (yield from plan)
            ^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/preprocessors.py", line 351, in run_wrapper
    yield from contingency_wrapper(plan,
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/preprocessors.py", line 606, in contingency_wrapper
    ret = yield from plan
          ^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/i22-bluesky/src/i22_bluesky/plans/linkam.py", line 133, in inner_linkam_plan
    yield from scan_linkam(
  File "/dls_sw/i22/software/blueapi/scratch/i22-bluesky/src/i22_bluesky/stubs/linkam.py", line 54, in scan_linkam
    yield from fly_and_collect(flyer)
  File "/venv/lib/python3.11/site-packages/dls_bluesky_core/stubs/flyables.py", line 51, in fly_and_collect
    yield from bps.wait(group=complete_group, timeout=flush_period)
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/plan_stubs.py", line 504, in wait
    return (yield Msg('wait', None, group=group, timeout=timeout))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/preprocessors.py", line 251, in msg_mutator
    _s = yield msg
         ^^^^^^^^^
  File "/dls_sw/i22/software/blueapi/scratch/bluesky/bluesky/run_engine.py", line 2277, in _status_object_completed
    raise FailedStatus(ret) from exc
bluesky.utils.FailedStatus: <AsyncStatus errored>
```

Seems to be an issue with the `prepare` protocol in how it calls `controller.arm`.

rosesyrett, 2024-01-24:
I've now looked into this, and discovered the source of the problem.

When we ran the I22 experiment in november, we discovered that if we tried to set `num=0` into all detectors, that the detectors would understand they need to take images forever. This turned out to not be the case for the Pilatus, so we modified the definition of a Pilatus (https://github.com/bluesky/ophyd-async/blob/main/src/ophyd_async/epics/areadetector/controllers/pilatus_controller.py) so that it set the number of triggers to 999,999.

The behaviour then was, we create the `Flyer` and prepare it, which means we arm all the detector controllers: https://github.com/bluesky/ophyd-async/blob/74edf2433e0b718c8c0d187994bb39e281539202/src/ophyd_async/core/flyer.py#L112-L117

By default, in the above code snippet, `num=0` would be passed to all the detector controllers for this process. The tetramm in the directory_provider branch of dodal would raise an Exception if it ever got something that wasn't num=0: https://github.com/DiamondLightSource/dodal/blob/directory_provider/src/dodal/devices/tetramm.py#L130-L131

Now, we have changed the logic such that the number of frames comes from the plan, and such that num=0 is no longer passed. This was done in https://github.com/bluesky/ophyd-async/pull/96, however we failed to consider that the Tetramm would now raise this exception, as now a `num` was being passed to it.

A PR has been raised for the dodal repository, where this issue really ought to belong. But at the time of testing this out on Tuesday, it seemed like an ophyd_async issue.

Perhaps we ought to reconsider, at a higher level, if the PR I've proposed is the right thing to do (i.e. get the tetramm to ignore `num` or if we need to rethink how controllers work. Maybe we need a new subclass of controllers, e.g. ContinuousControllers which don't care about number of frames and DiscreteControllers which do?

rosesyrett, 2024-01-24:
As an aside, we also need to consider the changes needed to the linkam plan in the i22-bluesky repository, now that the number of frames actually come from the plan (and is not assumed to be 0). I'm hoping that the changes are minimal but this needs further discussion.

coretl, 2024-06-17:
Seems no longer relevant

closed 2024-06-17 by coretl