bluesky/ophyd-async #115: Create a test to reproduce connection failures

DominicOram, 2024-02-19:
In Hyperion we are seeing random `ophyd-async` reconnect issues https://github.com/DiamondLightSource/hyperion/issues/1159. It would be good to get to the bottom of these. 

## Acceptance Criteria
* We write a dummy application that periodically reconnects to ophyd-async devices on i03 and try and reproduce the error. (We do not want to do this in a tight loop and steal all the resources, also probably best ran supervised on a beam off day)
* We try and find a pattern that can help us with diagnosing the issue

coretl, 2024-06-17:
Connections are now cached in #294

coretl, 2024-07-19:
@DominicOram is this still an issue?

DominicOram, 2024-07-22:
i don't think so. Happy to close and re-open if we see it again

closed 2024-07-22 by DominicOram