bluesky/ophyd-async #122: Consider Path Signals

callumforrester, 2024-02-23:
Some string PVs are intended only to store file paths, e.g. the HDF5 plugin write path. Perhaps there is value in representing these with a `Signal[Path]` rather than a `Signal[str]` and performing some validation on input values.

See [ophyd_async.epics.areadetector.writers.nd_file_hdf.NDFileHDF](https://github.com/bluesky/ophyd-async/blob/641894d6404072f4e3413a82383ef257186d0735/src/ophyd_async/epics/areadetector/writers/nd_file_hdf.py#L19) for an example.

@coretl any thoughts?

DominicOram, 2024-02-23:
:+1: from me

coretl, 2024-02-26:
Could be done, but I'd say this is low priority compared with the other things we're doing

callumforrester, 2024-02-26:
Will leave it as a good first issue then

coretl, 2024-07-22:
Closing as `PathProvider` makes the above usecase less compelling. We can revisit after 1.0 if this is still needed

closed 2024-07-22 by coretl