bluesky/ophyd-async #123: Add step scan support into the prepare for the standard detector

abbiemery, 2024-02-28:
This is a follow on from #116 . The prepare method deals with the fly scanning behaviour of the detector, but not the step scanning. This is not major issue as the prepare is not currently widely used i step scanning, but it could be. As prepare is meant to be able to deal with both a reasonable behaviour for `prepare` in the standard detector would be to have prepare set trigger values for both options.  It then will go on to perform its other tasks for fly scans. The step scan `trigger` method can check that the cached trigger info from prepare is for a software trigger (so not a flyscan) if it is to use it. If there is no trigger info then prepare has not been called and we should use the existing defaults ( as they are present now), if it does exist but its fly scanning trigger info, fail with an error.

jwlodek, 2024-03-01:
One thought: if #129 is true, then wouldn't `prepare` become required to be run after `stage` and before `trigger` even in a step scan, because otherwise the `writer` will never be opened. In that case, should `trigger` just fail outright if `self._trigger_info` is `None` (indicating `prepare` was not run)?

coretl, 2024-03-04:
I think that depends on whether we want to support doing step scans of `Preparable` detectors using existing plans. If yes, then we need to lazily call `prepare()` if not already done in `trigger()`. In not, then we can do as you say and just fail if it hasn't been run

coretl, 2024-03-04:
@danielballan @tacaswell thoughts on whether we want people to be able to upgrade to ophyd-async without changing their plans at all?

coretl, 2024-06-17:
I think we need a lazy prepare in `trigger()`

coretl, 2024-06-24:
Close in favour of #314

closed 2024-06-24 by coretl