bluesky/ophyd-async #127: Rename PandA to HDFPandA deriving from PandABlocks and StandardDetector

abbiemery, 2024-02-29:


abbiemery, 2024-03-07:
This makes sense to follow your work  @evalott100 so if you have time following yours that would be great. If not we'll roll it over but it's probably best to avoid someone else interfering.

evalott100, 2024-04-10:
There's a decision I'm trying to make...

Our HDFPandA is going to need take a controller on initialisation:

https://github.com/bluesky/ophyd-async/blob/f58a588f4b73ec7c00e1b81f980f7679d14713db/src/ophyd_async/panda/hdf_panda.py#L9-L17

However the controller itself will require blocks which are initialised at `HDFPandA.connect` time:

https://github.com/bluesky/ophyd-async/blob/f58a588f4b73ec7c00e1b81f980f7679d14713db/src/ophyd_async/panda/panda_controller.py#L44-L46

I've considered making a new abstract class, subclasses of which will contain a function to fill their attributes with the pvi devices after the pvi device has connected:

https://github.com/bluesky/ophyd-async/blob/f58a588f4b73ec7c00e1b81f980f7679d14713db/src/ophyd_async/epics/pvi/pvi.py#L302-L307

In our case, in the controller we have:

https://github.com/bluesky/ophyd-async/blob/f58a588f4b73ec7c00e1b81f980f7679d14713db/src/ophyd_async/panda/panda_controller.py#L24-L25

https://github.com/bluesky/ophyd-async/blob/f58a588f4b73ec7c00e1b81f980f7679d14713db/src/ophyd_async/panda/panda_controller.py#L54-L56

and in HDFPandA we have:

https://github.com/bluesky/ophyd-async/blob/f58a588f4b73ec7c00e1b81f980f7679d14713db/src/ophyd_async/panda/hdf_panda.py#L28-L33

This seems quite ugly to me, however I like the idea of having a way to see that a device will be given signals by a pvi device after it has been connected.

@coretl , @danielballan , @abbiemery , @callumforrester any ideas?

evalott100, 2024-04-10:
Why not something like:

``` python
T = TypeVar("T", bound=Signal)

class ConnectTimeSignal(Generic[T]):
    connected = False

    def __init__(self, signal: Optional[T] = None):
        self.signal = signal
        
    def connect(self, signal: T):
        self.signal, self.connected = self.signal, True

    def __call__(self) -> Optional[T]:
        return self.signal
        
class PcapBlock(Device):
    active: ConnectTimeSignal[SignalR[bool]]
    arm: ConnectTimeSignal[SignalRW[bool]]

class CommonPandABlocks(Device):
    status: SignalRW # some top level signal, e.g PANDA:STATUS
    pcap: PcapBlock
    
class HDFPanda(StandardDetector, CommonPandaBlocks):
    def __init__(self, prefix):
        self._backends = create_devices_from_annotations(self)
        # Same as yours except all the signals are initialised with ConnectTimeSignal(signal=None)
        super().__init__(
            controller=PandaPcapController(pcap_reference=self.pcap),
            # same for writer
        )
    async def connect(self, sim, timeout):
        await fill_devices_from_pvi(self, self._backends, self._prefix + "PVI")
        # self.pcap has some extra ConnectTimeSignal that were in pvi but not in type hints
````

This has a couple of benefits in my opinion:
    * It's explicit while reading, and at runtime what you're getting from PVI
    * You don't have to set up signals which are always going to be disconnected

coretl, 2024-04-10:
That's an interesting idea, but the major problem I can think about is that the `PcapBlock` looks different from every other `Device`, so you have to do `await self.pcap.active().get_value()` instead of `await self.pcap.active.get_value()`. I think the additional brackets is enough to mean we shouldn't do this, as it seems to infer you are calling the signal, and it is changing the public API of the Device to support an internal implementation detail...

evalott100, 2024-04-10:
Hmmm the only other thing I can think if is something like this where the wrapper is hidden away with `__getattribute__` on the controller but the pvi parser has a way to bypass this:

It doesn't work with top level signals though since there's no way access the wrapper without the block.
 
``` python
class ConnectTimeSignal(Generic[T]):
    def __init__(self, signal: Optional[T] = None):
        self._signal = signal
        self._connected = signal is not None

    def _set_signal(self, signal: T):
        self._signal, self._connected = signal, True

    def _get_signal(self):
        if not self._connected:
            # raise RuntimeError("Signal not connected")
            print("signal not connected", end=": ")
        return self._signal

class PVIDevice:
    def __getattribute__(self, name: str):
        attr = super().__getattribute__(name)
        if isinstance(attr, ConnectTimeSignal):
            return attr._get_signal()
        return attr

    def reg_get(self, name):
        return super().__getattribute__(name)

class SubSubBlock(PVIDevice):
    some_other_sig: ConnectTimeSignal[SignalRW[bool]]

class PcapBlock(PVIDevice):
    active: ConnectTimeSignal[SignalR[bool]]
    arm: ConnectTimeSignal[SignalRW[bool]]
    some_other_block: SubSubBlock

class Controller:
    def __init__(self, pcap: PcapBlock, status):
        self.pcap = pcap
        self.status = status

    def print_stuff(self, called_from):
        print(f"{called_from}: self.pcap.active: {self.pcap.active}")
        print(f"{called_from}: self.pcap.arm: {self.pcap.arm}")
        print(
            f"{called_from}: self.pcap.some_other_block.some_other_sig: {self.pcap.some_other_block.some_other_sig}"
        )
        print(f"{called_from}: self.status: {self.status}")

    def do_more_stuff(self):
        self.pcap.arm += 1
        self.pcap.active += 1
        self.pcap.some_other_block.some_other_sig += 1

class CommonPVIDevice(PVIDevice):
    # a standard block
    pcap: PcapBlock
    status: ConnectTimeSignal[SignalRW[bool]]

def create_devices_from_annotations(device):
    assert issubclass(type(device), PVIDevice)
    device.pcap = PcapBlock()
    device.pcap.active = ConnectTimeSignal()
    device.pcap.arm = ConnectTimeSignal()

    device.pcap.some_other_block = SubSubBlock()
    device.pcap.some_other_block.some_other_sig = ConnectTimeSignal()

    device.status = ConnectTimeSignal()

def fill_pvi_entries(device):
    # Yay! We've got the signals through pvi
    # once they're set here the'll be accessed as normal
    # in the controller
    assert issubclass(type(device), PVIDevice)
    device.reg_get("pcap").reg_get("active")._set_signal(1)
    device.reg_get("pcap").reg_get("arm")._set_signal(1)
    device.reg_get("pcap").reg_get("some_other_block").reg_get(
        "some_other_sig"
    )._set_signal(1)
    device.reg_get("status")._set_signal(1)

class SomePVIDevice(PVIDevice):
    def __init__(self):
        create_devices_from_annotations(self)
        self.controller = Controller(pcap=self.pcap, status=self.status)
        self.controller.print_stuff("__init__")

    def connect(self):
        fill_pvi_entries(self)
        self.controller.print_stuff(called_from="connect")

device = SomePVIDevice()
device.connect()
device.controller.do_more_stuff()
print(f"values can change on either side: pcap.active: {device.pcap.active}")
print(f"values can change on either side: pcap.arm: {device.pcap.arm}")
print(
    f"values can change on either side: pcap.some_o:some_o_sig: {device.pcap.some_other_block.some_other_sig}"
)
```
## Output
```
signal not connected: __init__: self.pcap.active: None
signal not connected: __init__: self.pcap.arm: None
signal not connected: __init__: self.pcap.some_other_block.some_other_sig: None
__init__: self.status: None
connect: self.pcap.active: 1
connect: self.pcap.arm: 1
connect: self.pcap.some_other_block.some_other_sig: 1
connect: self.status: None
values can change on either side: pcap.active: 2
values can change on either side: pcap.arm: 2
values can change on either side: pcap.some_o:some_o_sig: 2
```

PR #185 (2024-04-23): Make HDFPandA StandardDetector
Closes #127 

Requires https://github.com/bluesky/ophyd-async/pull/198
closed 2024-04-23 by evalott100