bluesky/ophyd-async #130: Add multiplers to prepare

abbiemery, 2024-02-29:


jwlodek, 2024-03-01:
Should `TriggerInfo.num` remain the number of triggers for the detector instance itself, or should it now be the number of triggers for the slowest detector in the scan, since we can presumably get the number of triggers for each detector based on the multiplier?

coretl, 2024-03-04:
> Should `TriggerInfo.num` remain the number of triggers for the detector instance itself, or should it now be the number of triggers for the slowest detector in the scan, since we can presumably get the number of triggers for each detector based on the multiplier?
> As best as I can tell, it would be easier to use the baseline `num` for triggers, since the original trigger_info is coming from the higher level HardwareTriggeredFlyable class, with the same one being passed to each detector in `prepare`. I guess the question becomes should each `StandardDetector` have it's `TriggerInfo` updated to the actual `num`, or should we leave it as the baseline frequency and recalculate actual num triggers whenever needed with the multiplier.

I think it should be the number of triggers sent to the detector. We will use a PandA Pulse block to take the input pulse train and multiply it up, changing the livetime and deadtime too, so it makes sense for it to calculate a new `TriggerInfo` with this information and send it to `detector.prepare()`. The only reason the detector needs the multiplier is so that it can put the right information in the descriptor and StreamResource documents.

coretl, 2024-03-14:
Depends on #127

coretl, 2024-06-24:
Do this as part of #314

closed 2024-06-24 by coretl