bluesky/ophyd-async #134: Update the tutorial docs to not rely on dev install

abbiemery, 2024-03-05:
Currently if you follow the instructions on the tutorial having just installed ophyd-async (pip or otherwise) you cannot complete the tutorial. As this is a user tutorial this needs addressing to either include the required dependencies or change the tutorial.

callumforrester, 2024-04-15:
Reviewers: @abbiemery

Kezzsim, 2024-04-16:
A discovery I made in thoroughly walking through the bluesky docs is that currently the 'TRY NOW>>' button on https://blueskyproject.io/ does not work, presently it links to a site with an expired certificate.

[image]

I'll see if I can get this fixed as well

Kezzsim, 2024-04-16:
@abbiemery I've gone through the entire tutorial now in iPython using a new conda environment to test this but I'm having trouble reproducing the problem.

1. Create new conda environment that's clean and has nothing installed. `conda create -n bluesky-hackathon python=3.11`
2. Activate said conda environment. `conda activate bluesky-hackathon `conda activate bluesky-hackathon `
3. Install *ophyd-async*. `pip install ophyd-async`
4. Start an iPython session and run every step of the Bluesky tutorial available [Here](https://blueskyproject.io/bluesky/tutorial.html)

Can you provide additional context to what the environment might be where it's failing? And on which step of the tutorial?

abbiemery, 2024-04-17:
Where did step 4 come from? Something like this is what we need, but it's not explicitly stated in [this](https://blueskyproject.io/ophyd-async/main/user/tutorials/using-existing-devices.html) ophyd-async tutorial I don't think?

from step 5 I think we may have got mixed up on which tutorials :)

abbiemery, 2024-04-17:
What we need for this issue is to find out which dependencies from the dev list in [this](https://github.com/bluesky/ophyd-async/blob/main/pyproject.toml) are used for the tutorial and put an instruction in the tutorial to pip install them. 

For ones which are ONLY used in the tutorial and not for dev, then these should be removed from the dev list.

Kezzsim, 2024-04-17:
Ahh okay, in the kanban view I could not see what repository this was a part of (_I guess because the board view is not public_) so I thought it was the defacto BlueSky repo and tutorial.

I'll take a look with this in mind.

abbiemery, 2024-05-09:
I had a look at this and this is the list of dependencies needed to get the demo to work if you're just pip installing them. I'm missing something because the plots do not draw along, however I'll come back to it. The key point at the moment is that it depends on a pre-release of ophyd-async, so there isn't much point doing it way in advance of the release.

ophyd-async>0.3a2 bluesky ophyd matplotlib Ipython pyepics>=3.4.2 aioca

coretl, 2024-05-15:
> I had a look at this and this is the list of dependencies needed to get the demo to work if you're just pip installing them. I'm missing something because the plots do not draw along, however I'll [...]

I think this should end up being `ophyd-async[ca] matplotlib Ipython` after we ditch the `ophyd` examples from the tutorials

abbiemery, 2024-05-23:
I didn't know we were ditching the `ophyd` examples

PR #765 (2025-03-12): Make a demo extra for tutorial deps
Fixes #134

### Instructions to reviewer on how to test:
1. Follow instructions in first and second tutorials
2. See that scans work ok

### Checks for reviewer
- [ ] Would the PR title make sense to a user on a set of release notes
closed 2025-03-12 by coretl