bluesky/ophyd-async #139: Make _RBV helper externally visible

DominicOram, 2024-03-07:
The pattern of having `_RBV` at the end of a PV for the corresponding readback is common outside just AD. `ophyd-async` should provide a helper function for it. One already exists at https://github.com/bluesky/ophyd-async/blob/c42afcb1f801d5f3f6ae88fd6a7fa8e0c7043893/src/ophyd_async/epics/areadetector/utils.py#L11 but it feels weird for my device to depend on area detector just for this.

coretl, 2024-03-08:
How about we expose as `ophyd_async.epics.signal.epics_signal_rw_suffixed(datatype, write_pv, read_pv_suffix="_RBV")`?

Suggestions for a better name welcome...

DominicOram, 2024-03-08:
Works for me

olliesilvester, 2024-04-03:
Or `epics_signal_rw` can have a `read_suffix = ""` parameter

coretl, 2024-04-04:
@olliesilvester so you would then write:
```python
s1 = epics_signal_rw(prefix + "SingleReadWritePv")
s2 = epics_signal_rw(prefix + "ReadPv", prefix + "WritePv")
s3 = epics_signal_rw(prefix + "SuffixedPv", read_suffix = "_RBV")
```
vs
```python
s1 = epics_signal_rw(prefix + "SingleReadWritePv")
s2 = epics_signal_rw(prefix + "ReadPv", prefix + "WritePv")
s3 = epics_signal_rw_suffixed(prefix + "SuffixedPv", "_RBV")
```

I'm not sure which is clearer... @DominicOram what do you think?

DominicOram, 2024-04-04:
Haha, this is such a small issue to have so much discussion.

My motivation is to be able to call `epics_signal_rw_rbv(prefix + "SuffixedPV")` and not have to be writing `_RBV` everywhere (basically the motivation of the existing `ad_rw` function). So @olliesilvester's suggestion doesn't fulfil that, but it does make things arguably a bit cleaner if you have a different standard than `_RBV` so you could do both?

```python
def epics_signal_rw_rbv(prefix, read_suffix= "_RBV"):
    epics_signal_rw(prefix, read_suffix = "_RBV")
```

PR #221 (2024-04-19): Add SignalRW helpers
Fixes #139
closed 2024-04-19 by coretl