bluesky/ophyd-async #145: Change descriptor references to datakey

coretl, 2024-03-12:


coretl, 2024-05-03:
First task is done, but the other tasks still need doing, changing the title to match the new scope.

DiamondJoseph, 2024-05-03:
Aren't parts 2, 3 done too? SignalBackend now has

```python
    @abstractmethod
    async def get_datakey(self, source: str) -> DataKey:
        """Metadata like source, dtype, shape, precision, units"""
```

with implementations for CA, PVA

and Signal has

```python

    @_add_timeout
    async def describe(self) -> Dict[str, DataKey]:
        """Return a single item dict with the descriptor in it"""
        return {self.name: await self._backend.get_datakey(self.source)}
```

coretl, 2024-05-03:
still typing...

coretl, 2024-05-03:
Updated the task list to match, but now I'm not sure if this is the best way to do it, thoughts?

coretl, 2024-06-17:
Closing as we will support this via numpy structured data via #257

PR #153 (2024-05-03): Use DataKey instead of Descriptor
Closes #145 

Descriptor is overloaded with a DescriptorDocument, which contains many DataKeys. A DataKey is the key to a data field, describing how it should be read.
closed 2024-06-17 by coretl