bluesky/ophyd-async #156: Repository size

callumforrester, 2024-03-25:
This repository takes up 251MB at time of writing and took me several minutes to clone. Currently unsure why or if there is any unnecessary history that can be removed. @coretl any thoughts?

coretl, 2024-03-25:
Size of `gh-pages`? Once you've adopted copier you could delete all the branches and then squash merge them down...

callumforrester, 2024-03-25:
Okay, this now depends on #158

abbiemery, 2024-04-29:
#158 is now merged. What do I have to do now to reduce the size of these pages?

coretl, 2024-04-29:
- Checkout `gh-pages`
- Delete all the directories apart from `main` and the release tags
- Make sure that `switcher.json` only has directories that exist in it
- Delete `.doctrees/environment.pickle` from each directory
- Push
- Check the landing page for all versions works on github,io
- If that is all fine the make a local `gh-pages-backup` branch on `origin/gh-pages`
- Then squash `gh-pages` branch to a single commit and force push it
- Check again that landing page for all versions works on github.io
- Check a fresh clone is much smaller

abbiemery, 2024-05-02:
```Receiving objects: 100% (7809/7809), 7.64 MiB | 6.90 MiB/s, done.```

Done :)

closed 2024-05-02 by abbiemery