bluesky/ophyd-async #172: Update Zebra

stan-dot, 2024-04-04:
for @DiamondJoseph  to fill out

DominicOram, 2024-04-08:
What is required for this? We've recently done some work to convert the zebra to ophyd-async in dodal: https://github.com/DiamondLightSource/dodal/pull/382

DiamondJoseph, 2024-04-08:
Good catch, I'd forgotten about #382 and had naively found the couple of commits from last year of adding a basic Zebra before the repository was renamed. I think what we actually want is [a ZebraTriggerLogic analogous to the PandA one on i22](https://github.com/DiamondLightSource/i22-bluesky/blob/cba944bae3defc87afe640c77de36dd5bc825ddb/src/i22_bluesky/plans/fly_collect.py#L26), which may end up being moved into ophyd-async (I believe it's being refactored into a SequenceTableTriggerLogic?)

I need to talk to @coretl about the plan for ScanSpec-driven generic flying

DominicOram, 2024-04-08:
Ok, could we rename this issue/add a description to make it clear what the discussion needs to be and the acceptance criteria to either have that discussion or to implement the solution? It would be useful for us, whilst we are working on the zebra atm, to have some visibility over what direction you might want to take it in.

coretl, 2024-06-17:
I would be inclined to leave the zebra support in dodal. We're never going to touch the EPICS code, and hopefully most of the usage will switch to PandA when we switch to bluesky

closed 2024-06-17 by coretl