bluesky/ophyd-async #173: Factor StreamResource producing code into a separate class

danielballan, 2024-04-04:
This seems generally good, from a DRY point of view, and would specifically be helpful in managing the transition from old-style to new-style StreamResource

danielballan, 2024-04-04:
- [ ] Identify all classes that call `create_stream_resource`.
- [ ] Refactor the shared logic into a separate reusable class.
- [ ] Add conditional branch to this logic, that creates new-style or old-style StreamResource based on the version of `event_model` installed.

In `main` it's used only here: https://github.com/bluesky/ophyd-async/blob/b134436293b3885eb71781c7995f35a97fba54c6/src/ophyd_async/epics/areadetector/writers/_hdffile.py#L26-L36

But these PRs in progress will also use it:
#144 
#141

abbiemery, 2024-04-16:
The other issues that were needed for this refactor have now been merged. So there should be a AD one, a PandA one and a Sim one.

skarakuzu, 2024-04-22:
I see only two `_HDFFile` classes: one in `_hdffile.py` and the other one is in `panda_hdf_file.py` . Is there still an unmerged PR ?

danielballan, 2024-04-22:
It looks like #144 placed related code in `src/ophyd_async/sim/pattern_generator.py`.

coretl, 2024-05-02:
> It looks like [#144](https://github.com/bluesky/ophyd-async/pull/144) placed related code in `src/ophyd_async/sim/pattern_generator.py`.

It's the `HdfStreamProvider` class in that file

danielballan, 2024-05-08:
Per discussion during today's standup:

- As noted above, this should support old-style StreamResource with `root` and `resource_path` as well as new-style StreamResource with `uri`. It should use one or the other based on detecting the version of `event_model` installed.
- DLS needs old-style StreamResource only for the next month or so, so we don't need much flexibility/configurability there. We can hard-code `root` to be `"/"` and `resource_path` to be the remaining (relative) path.

Examples:

```
# old-style
root="/"
resource_path="some/path/to/file.h5"

# new-style
uri="file://{hostname}/some/path/to/file.h5"
```

where `{hostname}` might be `import socket; socket.gethostname()` or some configurable thing.

PR #429 (2024-07-04): 173 stream resource seperate classes
Fixes #173
closed 2024-07-04 by abbiemery