bluesky/ophyd-async #181: DeviceVector Naming Convention Does not Match Static Device Components

callumforrester, 2024-04-09:
The `DeviceVector` class gives us a more flexible, dynamic alternative to writing classes with many similar components, instead of 

```python
class Panda(Device):
    def __init__(self):    
        self.ttl_out_1: TTLOut = TTLOut(...)
        self.ttl_out_2: TTLOut = TTLOut(...)
        self.ttl_out_3: TTLOut = TTLOut(...)
        self.ttl_out_4: TTLOut = TTLOut(...)
```

We can simply write:

```python
class Panda(Device):
    def __init__(self):    
        self.ttl_out: DeviceVector[TTLOut] = DeviceVector({i: TTLOut(...) for in range(1, 5})
```

If `Panda` is instantiated with a `DeviceCollector` it and its components are automatically assigned names (accessible via the `".name"` attribute), but with the first example the names are delimited by underscores e.g. `"panda-01_ttl-out-1"` and in the second they are delimited by hyphens e.g. `panda-01-ttl-out-1`.

I _believe_ that the convention we want to aim for is hyphens but would need others to confirm (any ideas @coretl @evalott100 @danielballan?). Once estalished, it should be relatively simple to change.

coretl, 2024-04-09:
> I believe that the convention we want to aim for is hyphens but would need others to confirm 

Naming is set with https://github.com/bluesky/ophyd-async/blob/7be73e3ce0f79fca2e02f77f366792ee84586ed7/src/ophyd_async/core/device.py#L47-L59

If we use `DeviceCollector` then the root name will be a python identifier `panda01`, so the first example will be `panda01-ttl_out_1` and the second will be `panda01-ttl_out-1`. As long as we make the root name not include hyphens then you should be able to split on the hyphen and walk down the tree to find the named Device. Not that I think we should do that in any code, but it's a useful convention for debugging.

evalott100, 2024-04-09:
`"test_device-device_vector-1-device"` is much easier to understand than `"test-device-device-vector-1-device"`.

If we wanted to be strict with it we could enforce that `"-"` isn't allowed in passed in names, or swap it for `"_"` manually - no ambiguity when debugging names?

callumforrester, 2024-04-09:
From discussion: The two examples I provided are actually meant to be different. `DeviceVector` adds another layer of nesting to the tree to prevent conflicts with other members and allow names starting with int (not permitted by Python variable names), closing this as it is not a bug.

closed 2024-04-09 by callumforrester