bluesky/ophyd-async #184: Find out why codecov/project keeps hanging on PR's and stop it

abbiemery, 2024-04-09:
I imagine this is due to the concurrent limit on CI jobs, and if it doesn't retrieve it in a given time it never does. However it would be nice stop it happening if we could, and if not stop it being a requirement on merging that it is fetched.

abbiemery, 2024-04-09:
The codecov step is failing, but the CI still marks it as passing in the codeCI job. Then the status is waiting for something never sent. The below should fail the CI no?

```
Run codecov/codecov-action@v3
==> linux OS detected
https://uploader.codecov.io/latest/linux/codecov.SHA256SUM
==> SHASUM file signed by key id 806bb28aed779869
==> Uploader SHASUM verified (a66af85a0b56b2179d568067cf951e5b40c440b7cbffc305b6b08bcf9eb17dd8  codecov)
==> Running version latest
==> Running version v0.7.2
/home/runner/work/_actions/codecov/codecov-action/v3/dist/codecov -n 3.10/ubuntu-latest -Q github-action-3.1.6 -f cov.xml
[2024-04-09T16:16:42.752Z] ['info'] 
     _____          _
    / ____|        | |
   | |     ___   __| | ___  ___ _____   __
   | |    / _ \ / _` |/ _ \/ __/ _ \ \ / /
   | |___| (_) | (_| |  __/ (_| (_) \ V /
    \_____\___/ \__,_|\___|\___\___/ \_/

  Codecov report uploader 0.7.2
[2024-04-09T16:16:42.761Z] ['info'] => Project root located at: /home/runner/work/ophyd-async/ophyd-async
[2024-04-09T16:16:42.764Z] ['info'] -> No token specified or token is empty
[2024-04-09T16:16:42.862Z] ['info'] Searching for coverage files...
[2024-04-09T16:16:42.911Z] ['info'] => Found 1 possible coverage files:
  cov.xml
[2024-04-09T16:16:42.911Z] ['info'] Processing /home/runner/work/ophyd-async/ophyd-async/cov.xml...
[2024-04-09T16:16:42.915Z] ['info'] Detected GitHub Actions as the CI provider.
[2024-04-09T16:16:43.311Z] ['info'] Pinging Codecov: https://codecov.io/upload/v4?package=github-action-3.1.6-uploader-0.7.2&token=*******&branch=move-trigger-logic&build=8618642242&build_url=https%3A%2F%2Fgithub.com%2Fbluesky%2Fophyd-async%2Factions%2Fruns%2F8618642242&commit=f9b9d7bb3b2dfc7489efcb24f60e3[14](https://github.com/bluesky/ophyd-async/actions/runs/8618642242/job/23622542285#step:6:15)d6ceb96f7&job=Code+CI&pr=&service=github-actions&slug=bluesky%2Fophyd-async&name=3.10%2Fubuntu-latest&tag=&flags=&parent=
[2024-04-09T[16](https://github.com/bluesky/ophyd-async/actions/runs/8618642242/job/23622542285#step:6:17):16:43.484Z] ['error'] There was an error running the uploader: Error uploading to https://codecov.io: Error: There was an error fetching the storage URL during POST: 404 - {'detail': ErrorDetail(string='Unable to locate build via Github Actions API. Please upload with the Codecov repository upload token to resolve issue.', code='not_found')}
[[20](https://github.com/bluesky/ophyd-async/actions/runs/8618642242/job/23622542285#step:6:21)24-04-09T16:16:43.485Z] ['info'] Codecov will exit with status code 0. If you are expecting a non-zero exit code, please pass in the `-Z` flag
```

coretl, 2024-04-09:
Has the codecov action been upgraded recently? The latest release requires a token as a secret. I'll add a org level one to the org tomorrow and see if that helps. FYI @mrakitin

coretl, 2024-04-10:
Looks like https://github.com/codecov/codecov-action/issues/1359

If we update to copier we will get the v4 action anyway, so I will put the token in the org in preparation...

PR #186 (2024-04-10): Update codecov action to v4
Temporary test to check this fixes #184, the real fix will be to adopt the copier template
commit af695856 (2024-04-10): Update codecov action to v4 (#186)

Temporary test to check this fixes #184, the real fix will be to adopt the copier template
commit 1c5afc9f (2024-04-12): Update codecov action to v4 (#186)

Temporary test to check this fixes #184, the real fix will be to adopt the copier template
commit ec67ecf6 (2024-04-12): Update codecov action to v4 (#186)

Temporary test to check this fixes #184, the real fix will be to adopt the copier template
closed 2024-04-10 by abbiemery