bluesky/ophyd-async #199: Test helpers for assering value, reading and configuration

callumforrester, 2024-04-12:
[Dodal](https://github.com/DiamondLightSource/dodal) developers have produced various useful ways to test ophyd-async devices such as [`verify()`](https://github.com/DiamondLightSource/dodal/blob/6b43b48fcb388c43f919b7d0866fb44d8f8199ee/tests/devices/unit_tests/test_synchrotron.py#L201). Some of these should probably make it back into `main` and sit alongside other helper functions like `set_sim_signal`.

I think we need 3 helper functions:
- `assert_value(signal: SignalR[T], value: T)`
- `assert_reading(readable: Readable, reading: Reading)`
- `assert_configuration(readable: Readable, configuration: Configuration)`
 
## Acceptance Criteria
- Functions in ophyd-async
- Tests provided
- Docs provided where appropriate
- Issues raised in dodal to replace `verefy()` and any other old functions that can be superseded

callumforrester, 2024-04-12:
@coretl

coretl, 2024-04-15:
Possibly also want `assert_emitted` from #206

PR #226 (2024-04-29): 199 test helpers for assering value reading and configuration
This is to fix #199 and possibly #206 

Acceptance Criteria

- [x]  Functions in ophyd-async:
- assert_reading
- assert_configuration
- assert_value
- assert_emitted
- [x]  Tests provided
- [x]  Docs provided where appropriate
- [x]  Issues raised in dodal to replace verefy() and any other old functions that can be superseded

For #206 Need to document 2 ways:
- [x] Exercise the device directly, i.e. call device.read() and inspect the output. Should use helpers made in #199 
- This is already convered in test_sensor_reading_shows_va [... cut]
closed 2024-04-29 by Relm-Arrowny