bluesky/ophyd-async #205: Cannot launch IPython with matplotlib in devcontainer

callumforrester, 2024-04-15:
@OCopping to fill out logs

OCopping, 2024-04-15:
```
(venv) root@diamtl04:/scratch/ryi58813/ophyd-async# ipython --matplotlib=qt5
Python 3.9.19 (main, Apr 10 2024, 15:19:04) 
Type 'copyright', 'credits' or 'license' for more information
IPython 8.18.1 -- An enhanced Interactive Python. Type '?' for help.
[TerminalIPythonApp] WARNING | Eventloop or matplotlib integration failed. Is matplotlib installed?
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
File /venv/lib/python3.9/site-packages/IPython/core/interactiveshell.py:3645, in InteractiveShell.enable_matplotlib(self, gui)
   3641         print('Warning: Cannot change to a different GUI toolkit: %s.'
   3642                 ' Using %s instead.' % (gui, self.pylab_gui_select))
   3643         gui, backend = pt.find_gui_and_backend(self.pylab_gui_select)
-> 3645 pt.activate_matplotlib(backend)
   3646 configure_inline_support(self, backend)
   3648 # Now we must activate the gui pylab wants to use, and fix %run to take
   3649 # plot updates into account

File /venv/lib/python3.9/site-packages/IPython/core/pylabtools.py:368, in activate_matplotlib(backend)
    363 # Due to circular imports, pyplot may be only partially initialised
    364 # when this function runs.
    365 # So avoid needing matplotlib attribute-lookup to access pyplot.
    366 from matplotlib import pyplot as plt
--> 368 plt.switch_backend(backend)
    370 plt.show._needmain = False
    371 # We need to detect at runtime whether show() is called by the user.
    372 # For this, we wrap it into a decorator which adds a 'called' flag.

File /venv/lib/python3.9/site-packages/matplotlib/pyplot.py:342, in switch_backend(newbackend)
    339 # have to escape the switch on access logic
    340 old_backend = dict.__getitem__(rcParams, 'backend')
--> 342 module = importlib.import_module(cbook._backend_module_name(newbackend))
    343 canvas_class = module.FigureCanvas
    345 required_framework = canvas_class.required_interactive_framework

File /usr/local/lib/python3.9/importlib/__init__.py:127, in import_module(name, package)
    125             break
    126         level += 1
--> 127 return _bootstrap._gcd_import(name[level:], package, level)

File <frozen importlib._bootstrap>:1030, in _gcd_import(name, package, level)

File <frozen importlib._bootstrap>:1007, in _find_and_load(name, import_)

File <frozen importlib._bootstrap>:986, in _find_and_load_unlocked(name, import_)

File <frozen importlib._bootstrap>:680, in _load_unlocked(spec)

File <frozen importlib._bootstrap_external>:850, in exec_module(self, module)

File <frozen importlib._bootstrap>:228, in _call_with_frames_removed(f, *args, **kwds)

File /venv/lib/python3.9/site-packages/matplotlib/backends/backend_qt5agg.py:7
      4 from .. import backends
      6 backends._QT_FORCE_QT5_BINDING = True
----> 7 from .backend_qtagg import (    # noqa: F401, E402 # pylint: disable=W0611
      8     _BackendQTAgg, FigureCanvasQTAgg, FigureManagerQT, NavigationToolbar2QT,
      9     FigureCanvasAgg, FigureCanvasQT)
     12 @_BackendQTAgg.export
     13 class _BackendQT5Agg(_BackendQTAgg):
     14     pass

File /venv/lib/python3.9/site-packages/matplotlib/backends/backend_qtagg.py:9
      5 import ctypes
      7 from matplotlib.transforms import Bbox
----> 9 from .qt_compat import QT_API, QtCore, QtGui
     10 from .backend_agg import FigureCanvasAgg
     11 from .backend_qt import _BackendQT, FigureCanvasQT

File /venv/lib/python3.9/site-packages/matplotlib/backends/qt_compat.py:112
    108         raise AssertionError(f"Unexpected QT_API: {QT_API}")
    111 if QT_API in [QT_API_PYQT6, QT_API_PYQT5, QT_API_PYSIDE6, QT_API_PYSIDE2]:
--> 112     _setup_pyqt5plus()
    113 elif QT_API is None:  # See above re: dict.__getitem__.
    114     if _QT_FORCE_QT5_BINDING:

File /venv/lib/python3.9/site-packages/matplotlib/backends/qt_compat.py:82, in _setup_pyqt5plus()
     80     _to_int = operator.attrgetter('value')
     81 elif QT_API == QT_API_PYSIDE6:
---> 82     from PySide6 import QtCore, QtGui, QtWidgets, __version__
     83     import shiboken6
     84     def _isdeleted(obj): return not shiboken6.isValid(obj)

ImportError: libGL.so.1: cannot open shared object file: No such file or directory
```

callumforrester, 2024-04-15:
So, here is a repo with a working container that launches an IPython terminal with matplotlib: https://github.com/DiamondLightSource/htss-rig-bluesky

The main differences are in the Dockerfile and the settings to launch the container (which are in the README). You'll need to translate these settings from CLI arguments to the language of `devcontainer.json`

callumforrester, 2024-04-15:
Turns out it just works using TK as a backend, However may be a related issue: https://github.com/bluesky/bluesky/issues/1693

tacaswell, 2024-04-15:
You may run into other problems with tk.  This reads like there is something wrong with the pyside6 install.

https://github.com/bluesky/bluesky-pods/blob/main/compose/acq-pod/launch_bluesky.sh

has the magic needed to forward X11 out of a container to the desktop

callumforrester, 2024-04-15:
X11 forwarding seems to work for us out-of-box (i.e. if I just do `MPLBACKEND=tkagg python` and plot something it works fine. This error only happened with qt

tacaswell, 2024-04-15:
I misread as "containers locally" vs "containers on GH resources"