bluesky/ophyd-async #208: Create async versions of all the protocols we use in ophyd-async

coretl, 2024-04-15:
At the moment we import `from bluesky.protocols import Readable, Configurable, ...`. This has 2 issues:
- If we extend the protocol in ophyd-async (e.g. add `Collectable.get_index`) then we need a release of bluesky before we can use them
- If we type narrow (e.g. we don't have `SyncOrAsync` returns, we write `async def` instead) then the editor picks up the extended version when we override it, so we have to edit the text to make it match our narrowed version

I suggest we make `ophyd_async.protocols`, where any protocol that has `SyncOrAsync` in it is narrowed for use in ophyd-async. At least:
- `AsyncReadable`
- `AsyncConfigurable`
- `AsyncPausable`

then update any imports to use them.

This also gives us a place to put things like `Watchable` from #176 until they can migrate into bluesky

coretl, 2024-04-15:
For example, this:
https://github.com/bluesky/bluesky/blob/f03451a7e188df867df7fc86c70e3efe65a02709/src/bluesky/protocols.py#L263-L303

Would turn into this:
```python
from bluesky.protocols import HasName, Reading, DataKey, ...

@runtime_checkable
class AsyncReadable(HasName, Protocol):
    """Readable narrowed to be async only"""
    @abstractmethod
    async def read(self) -> Dict[str, Reading]:
        ...

    @abstractmethod
    async def describe(self) -> Dict[str, DataKey]:
        ...
```

callumforrester, 2024-04-15:
Reviewers: @coretl

callumforrester, 2024-04-19:
Fixed in #219

closed 2024-04-19 by callumforrester