bluesky/ophyd-async #213: Create class for the areaDetector statistics plugin.

jwlodek, 2024-04-15:
Several of our beamlines make extensive use of the statistics plugin to collect metadata (max pixel value, average value, etc.). We should add support for using this plugin with `ophyd-async`.

callumforrester, 2024-04-15:
Reviewers: @RobertSchaffer1 @DiamondJoseph

DiamondJoseph, 2024-04-15:
Here's my incredibly naive initial thoughts on it

https://github.com/bluesky/ophyd-async/pull/178

We want to be able to instantiate various AreaDetector implementations optionally with the Stats plugin (and likely other plugins in a chain later) and have an easy way of adding them to the scalar_signals of the detector (probably non-scalar too).

DiamondJoseph, 2024-07-09:
@jmaruland is this still ongoing?

ZohebShaikh, 2024-07-26:
closed by #468

closed 2024-07-26 by ZohebShaikh