bluesky/ophyd-async #239: Support enum signals for enum classes that are generated at runtime

DiamondJoseph, 2024-04-18:
As part of instantiating (e.g. a PandaA) from its PVI tree, we have a series of enumeration signals for which we do not know the possible values they can take. We should be able to instantiate those signals, and access a generated Enum class which allows iterating through all of the possible choices that that underlying PV supports. This type is not available for static type checking.

coretl, 2024-05-08:
Should use the `Literal` syntax defined in https://github.com/bluesky/ophyd-async/pull/249#issuecomment-2069609496

DiamondJoseph, 2024-05-09:
Isn't the Literal syntax for #238 and this is `GeneratedChoice`?

coretl, 2024-05-15:
> Isn't the Literal syntax for #238 and this is `GeneratedChoice`?

Yes, you're right. However I think we should drop `GeneratedChoices` as it's not pickleable, and instead treat these as `str` datatype, but put `choices` in the output of `get_descriptor`. This is still introspectable enough for us, while still being a simple datatype to pass around and serialize, and looks a lot like the output of #238

closed 2024-06-20 by evalott100