bluesky/ophyd-async #257: Include specific numpy dtype in output of `describe()`

danielballan, 2024-04-23:
As required by `event-model`, the `dtype` in `describe()` is a JSONschema data type, such as `"number"` or `"array"`. For Tiled (and perhaps for other consumers) we need to know that actual machine data type, such as `"<f8"` or `"<i4"`. The event-model does not currently specify a place to put this, but databroker+tiled have been using a new key `dtype_str` to place this (optionally) alongside the JSONschema `dtype`.

It has been observed this, coming fresh to `dtype` and `dtype_str`, it's not at all clear what these keys mean. We should evaluate our options, with a view toward both using obvious names and supporting backward-compatibility. Then we update `event-model` as needed and add the numpy-style data types here.

danielballan, 2024-05-30:
This design decision affects the schema, so I started an Issue at https://github.com/bluesky/event-model/issues/308. Once we decide on the spelling, we can implement `describe()` acordingly.

coretl, 2024-06-17:
- numpy types for arrays
- python types for scalars, but limit soft signals to int64 and float64

coretl, 2024-07-01:
Add datatype to `ShapeProvider` as part of this

coretl, 2024-07-09:
Decided on `dtype_numpy` instead of `dtype_str` over in https://github.com/bluesky/event-model/issues/308

abbiemery, 2024-07-11:
Closed with #438

closed 2024-07-11 by abbiemery