bluesky/ophyd-async #259: The docs are now worse?

abbiemery, 2024-04-24:
I can't tell from the existing issues if there is visibility on this, but the docs are now much worse than before we changed the template.

Templates are visible, things are missing.
[image]

This is worse to navigate.
[image]

The dev instructions have gone?

joeshannon, 2024-04-29:
For the `:::` it looks like it needs an extension enabling:

https://github.com/DiamondLightSource/blueapi/blob/c3d1ccd89dd22725931b628c857f3c953d2e3486/docs/conf.py#L55

In the current copier template the Developer and User guides have been merged. The other docs such as running tests etc. are now linked directly to the copier template from how-to/contribute.html.

abbiemery, 2024-05-01:
> In the current copier template the Developer and User guides have been merged.

But, why? There are sometimes very good reasons to have these separate. I know this isn't an argument because the decision has been made but this feels like a step back to me.

coretl, 2024-05-01:
> > In the current copier template the Developer and User guides have been merged.
> 
> But, why? There are sometimes very good reasons to have these separate. I know this isn't an argument because the decision has been made but this feels like a step back to me.

The decision was made because there is a blurry boundary between developer and user guides, and feedback was that people ended up missing docs because they weren't reading the right section. Most of the developer guide has now been put in the copier template docs and linked from the repo `how-to/contribute` guide. Suggestions on what should be in this guide welcome.

PR #272 (2024-05-01): Fix docs and tidy README
Fixes #259
closed 2024-05-01 by coretl