bluesky/ophyd-async #260: Change from using `asyncio.wait_for(x())` to `async with asyncio.timeout(timeout: float): await x()`

evalott100, 2024-04-24:
Some interesting asyncio behaviour was discovered by myself and @coretl today while debugging test failures:

The following tests seem roughly equivalent, however...

``` python

# THIS PASSES
async def test_wait_for_in_coroutine():
    async def meh():
        await asyncio.wait_for(asyncio.sleep(10), timeout=5)

    task = asyncio.create_task(meh())
    task.cancel()
    with pytest.raises(asyncio.CancelledError):
        await task

# THIS FAILS
async def test_sleep_in_coroutine():
    async def meh():
        await asyncio.sleep(10)

    task = asyncio.create_task(asyncio.wait_for(meh(), timeout=5))
    task.cancel()
    with pytest.raises(asyncio.CancelledError):
        await task

# THIS PASSES
async def test_timeout_in_coroutine():
    async def meh():
        async with asyncio.timeout(5):
            await asyncio.sleep(10)

    task = asyncio.create_task(meh())
    task.cancel()
    with pytest.raises(asyncio.CancelledError):
        await task

# THIS FAILS
async def test_wait_for_in_task_init():
    task = asyncio.create_task(asyncio.wait_for(asyncio.sleep(10), timeout=5))
    task.cancel()
    with pytest.raises(asyncio.CancelledError):
        await task

# THIS PASSES
async def test_wait_for_in_task_init_sleep_zero():
    task = asyncio.create_task(asyncio.wait_for(asyncio.sleep(10), timeout=5))
    await asyncio.sleep(0)
    task.cancel()
    with pytest.raises(asyncio.CancelledError):
        await task
```

When the task is cancelled, it's possible for the further nested coroutines to remain active (we think, but aren't sure). This isn't even a special property of `wait_for` but is a problem inherent in asyncio that presents itself when pytest tries to teardown:

``` python
# THIS FAILS
async def test_wait_for_sleep_in_coroutine_1():
    async def meh1():
        await asyncio.sleep(10)

    async def meh2(coro):
        await coro

    task = asyncio.create_task(meh2(meh1()))
    task.cancel()
    with pytest.raises(asyncio.CancelledError):
        await task

# THIS PASSES
async def test_wait_for_sleep_in_coroutine_2():
    async def meh1():
        await asyncio.sleep(10)

    async def meh2(coro):
        await coro

    task = asyncio.create_task(meh2(meh1()))
    await asyncio.sleep(0)
    task.cancel()
    with pytest.raises(asyncio.CancelledError):
        await task
```

@dperl-dls apparently you were having similar issues...

In particular, this line was causing a very unobvious failure in the teardown:

https://github.com/bluesky/ophyd-async/blob/4c8ce613d01d76ca196b5f6e3540bb3560cf15e2/src/ophyd_async/core/sim_signal_backend.py#L153-L154

changing it to

``` python
        if wait:
            async with asyncio.timeout(timeout):
                await self.put_proceeds.wait()
```

did the trick.

evalott100, 2024-04-25:
Okay I've found a much simpler way to deal with this.

If we create a task/event locally, we add it to a global set

https://github.com/bluesky/ophyd-async/blob/a334baaf53039ba8506de5f460cd64ddbba549e5/src/ophyd_async/core/async_status.py#L11-L13

https://github.com/bluesky/ophyd-async/blob/a334baaf53039ba8506de5f460cd64ddbba549e5/src/ophyd_async/core/signal.py#L252-L254

Then we get remove them from the queue when we're done

https://github.com/bluesky/ophyd-async/blob/a334baaf53039ba8506de5f460cd64ddbba549e5/src/ophyd_async/core/async_status.py#L29-L32

https://github.com/bluesky/ophyd-async/blob/a334baaf53039ba8506de5f460cd64ddbba549e5/src/ophyd_async/core/signal.py#L273-L288

The sets aren't used at any point, but they allow the garbage collector to properly dispose of their contents rather than the event loop destroying them flakily unreliably.

coretl, 2024-04-29:
Bear in mind that this is a quick fix, but it is masking valid errors. If tasks are spawned and not tracked then if they error then the tracebacks will be unreported.

Long term we need to take this out, then either replace with `async with asyncio.timeout` or targetted callback tracking like this but for each individual case...

dperl-dls, 2024-05-07:
Unfortunately `asyncio.timeout` is only available in python 3.11+

evalott100, 2024-05-08:
I've tried to play around with an `AsyncStatus.__del__` to kill tasks on teardown only if we're in a pytest environment. This hasn't been very fruitful... lots of event loop behind the scenes magic.

I think we could be a little hacky for now and only add to the global sets if we're in a pytest run? We could make methods `make_task_visible_to_pyytest_teardown`, etc., and hide the global set in a file with them.

This means:
    * In pytest, the tasks which would close correctly, but too late, won't give us warnings
    * In production we'll get the errors we want

coretl, 2024-05-08:
Personally I'd prefer to track down the errors rather than do this. Are they causing CI failures at the moment?

evalott100, 2024-05-09:
> Personally I'd prefer to track down the errors rather than do this. Are they causing CI failures at the moment?

They don't without #251

closed 2024-08-13 by evalott100