bluesky/ophyd-async #261: Simulated backend should check enums inherit from strings

DominicOram, 2024-04-24:
If I accidentally do:
```python

class MyEnum(enum):
     test = "TEST"

class MyDevice(StandardReadable)
    def __init__():
         self.signal = epics_signal_rw(MyEnum, "blah")
```

I can run:

```python
my_device = MyDevice()
await my_device.connect(sim=True)
```
and I get no errors. However, if I run it in production I will get an error saying `signal: TypeError: blah has type Enum but doesn't inherit from String`. It would be nicer to catch this class of errors in my tests rather than in production and so the simulated backend should throw a similar error

DiamondJoseph, 2024-05-20:
Note that they should also check the other requirements of signal backends- if possibly they should use the same handling as in backends/_common.py

callumforrester, 2024-05-20:
Closing in favour of #310, which provides the check for all backends