bluesky/ophyd-async #266: Initial values on soft signals not carried through to sim

DominicOram, 2024-04-29:
As a developer I would expect the initial values on my soft signals to be the same regardless of whether I connect with `sim` or not.
i.e. I have a device:
```python

from ophyd_async.core import (
    StandardReadable,
    soft_signal_rw,
)

class MyDevice(StandardReadable):

    def __init__(self, prefix: str, name: str = ""):
        self.my_signal= soft_signal_rw(int, 10)
```

and the following test fails:
```python
async def test_init() -> MyDevice:
    device = MyDevice("-DI-OAV-01", name=DEVICE_NAME)
    await device.connect(sim=True)

    assert await device.my_signal.get_value() == 10
```
However, running it as a real device passes:
```python
async def test_init() -> MyDevice:
    device = MyDevice("-DI-OAV-01", name=DEVICE_NAME)
    await device.connect()

    assert await device.my_signal.get_value() == 10
```

Is this expected behavior? It is unexpected for my usecase of trying to test the real device as much as possible

coretl, 2024-04-29:
This is not desired behaviour, although I can see how it might be happening at the moment.

@evalott100 please can you check this will be fixed when https://github.com/bluesky/ophyd-async/issues/94 is fixed?

evalott100, 2024-05-07:
I've fixed this  in #251:

The intial value is now correctly passed through to the mock backend post connect:

https://github.com/bluesky/ophyd-async/blob/f9bf5a98856bf75d181da80fdd99cc39df62c2e1/tests/core/test_mock_signal_backend.py#L308-L323

closed 2024-05-10 by coretl