bluesky/ophyd-async #267: Writing to soft signals fails in sim

DominicOram, 2024-04-29:
With the new `soft_signal_r_and_backend` trying to write to a device fails in sim mode. i.e. if I have the following device:

```python
class MyDevice(StandardReadable):
    def __init__(self, prefix: str, name: str = ""):
        self.signal, self._signal_backend= soft_signal_r_and_backend(int)
   
    def set(self, _):
         await self._signal_backend.put(1)
```

The following test fails with the exception of ` NotImplementedError: No PV has been set as connect() has not been called`:

```python
async def test_set():
    device = MyDevice("-DI-OAV-01")
    await device.connect(sim=True)
    await device.set(0)
```

This is presumably because the signal backend gets swapped out for the sim back end but `_signal_backend` never gets updated. This makes testing soft signals very hard.

coretl, 2024-04-29:
This also looks like a bug. Writing to soft signals should work the same whether in sim or real mode. This should be fixed in #94, where the only difference between connecting a soft signal in mock mode is that calls to `backend.put` will go via a mock, so will be captured, as well as storing the value on the signal so `signal.get_value` works the same.

@evalott100 is my understanding correct?

evalott100, 2024-05-07:
I've fixed this in the linked PR:

https://github.com/bluesky/ophyd-async/blob/f9bf5a98856bf75d181da80fdd99cc39df62c2e1/src/ophyd_async/core/signal.py#L264-L279

The method has been changed to give a callable that will put directly - this allows the same function to work even if the backend has been swapped out with a `MockSignalBackend` post connect.

evalott100, 2024-05-07:
I've also included a test:

https://github.com/bluesky/ophyd-async/blob/f9bf5a98856bf75d181da80fdd99cc39df62c2e1/tests/core/test_mock_signal_backend.py#L326-L344

closed 2024-05-10 by coretl