bluesky/ophyd-async #280: Make p4p backend allow using `int` datatypes for `float` PVs only if `PREC` is 0

evalott100, 2024-05-02:
In https://github.com/bluesky/ophyd-async/pull/276, we fixed the bug that the panda intialises an analogOut with precision 0 as an int. The signal is made (correctly) with an `int`, however `make_converter` fails

https://github.com/bluesky/ophyd-async/blob/58e1cf47b6895093dd029c941f42362565d5a555/src/ophyd_async/epics/_backend/_p4p.py#L219-L220

since `float` is not a subclass of `int`.

The fix in in https://github.com/bluesky/ophyd-async/pull/276 allows the signal to be made anyway. We should add the constraint that `.PREC == 0`.

evalott100, 2024-05-07:
Once  the following are merged we can drop the changes introduced in #276

https://github.com/PandABlocks/PandABlocks-ioc/issues/112
https://github.com/dls-controls/pythonSoftIOC/issues/160

We've decided these kind of checks shouldn't be handled in ophyd-async - pandablocks-ioc should be using an int for the record if ophyd-async expects an int datatype.

coretl, 2024-06-04:
This has since come up with the CA backend, so I think we need to support this anyway. I suggest we:
- Make the changes from #276 in the CA converter too
- Also assert `value.precision == 0` in the CA converter
- Also assert `value.display.precision == 0 ` in the PVA converter
- Add records in `test_records.db` and a test in `test/epics/test_signals.py` to check

evalott100, 2024-08-05:
> are you doing this? Otherwise MX-DAQ would be happy to pick up

That would be much appreciated, there are a few tickets I'll be doing in this sprint before I get round to this one.

PR #509 (2024-08-13): Allow precision 0 float records to be represented as int signals
Fixes #280
closed 2024-08-13 by evalott100