bluesky/ophyd-async #297: PVI connection handling unable to understand forward referencing in container

DiamondJoseph, 2024-05-10:
NB. Posisbly related to blueapi's subprocess handling and pickling of objects/types.

Frustrating issue when trying to connect to a PandA that get_type_hints was unable to resolve the forward reference in Device.parent -> Optional[Device]

evalott100, 2024-05-10:
For posterity:

We found that this was a specific problem related to `future.__annotations__` alongside `get_type_annotations`. For whatever reason, in the blueapi build of ophyd-async, we would get an error where `Optional[Device]` (normally unpacked to `Union[None, Device] at runtime) was not defined, after taking out the optional we had the error that `Device` wasn't defined.

Since the behaviour of `future.__anotations__` is default in python > 3.10/3.11, we don't need this anyway.

coretl, 2024-05-15:
Eventually this needs sorting, as annotations will all be like this in the end, but fortunately there is no set date for this [at the moment](https://docs.python.org/3/library/__future__.html#id2) so fine with this for now

olliesilvester, 2024-05-20:
We use `from __future__ import annotations` lots in Hyperion, so we aren't able to use the Panda anymore until this is fixed

PR #298 (2024-05-20): Replace forward ref handling for Device
Fixes #297
closed 2024-05-20 by DiamondJoseph