bluesky/ophyd-async #308: Decide how to make `DetectorControl.get_deadtime` read data from PVs

coretl, 2024-05-15:
We want `DetectorControl.get_deadtime` to be sync so that we can call it in plans while the detector is doing something else, but we also want to be able to depend on various PVs (like frame depth and model number). One idea is to call `get_deadtime` with the output of `read_configuration`, either live during the plan, or using cached values from the saved config so it can be done during the plan stub. This still leaves read only fields like model number, but we could add a `DetectorControl.connect` method that could get that.

_Originally posted by @evalott100 in https://github.com/bluesky/ophyd-async/pull/305#discussion_r1598427360_

coretl, 2025-02-25:
We decided on a simpler solution:
- `get_deadtime(exposure)` is a safe minimum deadtime that is always valid in any mode of the detector
- If we pass a smaller deadtime than that, we should validate it with the settings of the detector at runtime

Still need to think about how the clock skew can be handled here, maybe `get_deadtime` needs to know if it's getting a gate or a trigger so it knows what to adjust for the crystal accuracy?

olliesilvester, 2025-07-29:
The Jungfrau detector (allegedly) has an internal deadtime of 0, but it also has a settable `period_between_frames` signal in fastcs. It would be nice for `get_deadtime` to be able to read this signal, but that requires it being an async function

In the meantime, I am planning to make a `create_jungfrau_trigger_info` function which makes sure 
`TriggerInfo.livetime = exposure_time - period_between_frames`. Then `get_deadtime` can always return 0