bluesky/ophyd-async #311: SoftSignal: Add a default Converter for dataclasses or pydantic objects

dperl-dls, 2024-05-16:
When we have these, it's probably nice to convert them to json by default.

callumforrester, 2024-05-20:
I think that sounds sensible.
If a signal returns a pydantic model/dataclass/dictionary, what do we expect the documents to look like?

dperl-dls, 2024-05-20:
Something where I can do:

```python
reading = doc["data"]["my_device-my_pydantic_signal"]
result = MyPydanticSignalType(**reading)
do_something_with(result.field1)
do_something_else_with(result.field2)
```

So basically, the reading data should IMO be the `dict`/`json` representation of the thing

coretl, 2024-05-23:
I think this sounds reasonable:
- `Signal.get_value` returns the native datatype, which could be Enum, pydantic model, dataclass, numpy array, etc.
- `Signal.get_reading` return the serializable dictionary representation of it
The only exception I would like to make is numpy arrays, I would like to treat these as primitives rather than turning them into python lists in get_reading as we do this conversion any time we get a subscription update, so this would be too costly

@dperl-dls do numpy arrays serialize fine when sent over zmq?
@callumforrester do numpy arrays serialize fine in pydantic?

callumforrester, 2024-05-23:
@coretl yes but with extra steps: https://github.com/pydantic/pydantic/issues/7017

callumforrester, 2024-05-23:
Another consideration: Should `get_value` return a value or a reference? I'm a bit apprehensive about something like:

```python
my_dict = await device.soft_signal.get_value()
assert my_dict["foo"] == 5

my_dict["foo"] = 6

my_dict = await device.soft_signal.get_value()
assert my_dict["foo"] == 6
```

As it might be confusing.

coretl, 2024-05-23:
Good point, maybe we use something like [frozendict](https://pypi.org/project/frozendict/#deepfreeze-api) to return from `get_value`?

coretl, 2024-06-24:
> I think this sounds reasonable:
>
>     * `Signal.get_value` returns the native datatype, which could be Enum, pydantic model, dataclass, numpy array, etc.
>
>     * `Signal.get_reading` return the [...]

Decided to have both `get_value` and `get_reading` return the native datatype, then do the conversion in the JSON/YAML/MsgPack/whatever layer

coretl, 2024-07-22:
To be done as part of #310

closed 2024-07-22 by coretl