bluesky/ophyd-async #314: Defer creation of HDF5 file until prepare/trigger

callumforrester, 2024-05-17:
Calling in `prepare`/`trigger` rather than `stage` would mean we can tie the file path into contextual information about the run. `stage` is normally called before `open_run` so creating the file in `stage` means the run does not exist at the time of creation. 
Any solution has to be mindful of pre-declare, the file must exist by the time `describe` is called, i.e. the `declare_stream` plan.

abbiemery, 2024-06-11:
This is needed so we can use multipliers. Closing old issue #129 in favour of this.

coretl, 2024-06-24:
Suggested changes to `StandardDetector`:
- `stage()` brings the device into a state that is ready to start *any* scan, no matter of its parameters
- `prepare()` will set multipliers, and get a filename, and call `writer.open()`
- if `prepare()` has not been called with a software trigger mode before `trigger()`, then call it for the user with default parameters

Should be done with knowledge of how the interface between `StandardDetector` and `DetectorController`: https://github.com/bluesky/ophyd-async/issues/405

closed 2024-07-26 by ZohebShaikh