bluesky/ophyd-async #319: Errors in `MockSignalBackend` `callback`/`side_effect` are not reported.

dperl-dls, 2024-05-20:
If you have something like:

```python
from ophyd_async.epics.motion import Motor
from ophyd_async.core import (callback_on_mock_put)

def patch_motor(motor: Motor, initial_position=0):
    return callback_on_mock_put(
        motor.user_setpoint, lambda v, **_: set_mock_value(motor.user_readback, v)
    )

async def test_patched_motor():
    m = Motor("")
    await m.connect(mock=True)
    with patch_motor(m):
        await m.set(10)
        assert await m.user_readback.get_value() == 10
```

This will fail, because the implicit kwargs `wait` and `timeout` from `.set()` are passed to the lambda, which doesn't take them, but this failure is not reported, and the callback simply doesn't happen.

If you instead do:

```python
def patch_motor(motor: Motor, initial_position=0):
    return callback_on_mock_put(
        motor.user_setpoint, lambda v, **_: set_mock_value(motor.user_readback, v)
    )
```

then there is no problem.

This is quite mysterious and could be at least better documented in the docstring and maybe type bound of `(T@callback_on_mock_put) -> None`, and it would be better if the exception from the mock was propagated to the user.

evalott100, 2024-05-22:
I'm a little confused by this...

The following test will fail

``` python

async def test_callback_on_mock_put_fails():
    mock_signal = SignalRW(SoftSignalBackend(float))
    await mock_signal.connect(mock=True)

    def some_function_without_kwargs(arg):
        pass

    callback_on_mock_put(mock_signal, some_function_without_kwargs)
    await mock_signal.set(10.0)
```

with 

```
TypeError: test_callback_on_mock_put_fails.<locals>.some_function_without_kwargs() got an unexpected keyword argument 'wait'
```

and when I try with a lambda I get

```
TypeError: test_callback_on_mock_put_fails.<locals>.<lambda>() got an unexpected keyword argument 'wait'
```

evalott100, 2024-05-22:
Changing this to a `SimMotor` (to prevent a 0 velocity error), I can confirm that the following fails as it should:

``` python
def patch_motor(motor: Motor, initial_position=0):
    return callback_on_mock_put(
        motor.user_setpoint, lambda v: set_mock_value(motor.user_readback, v)
    )

async def test_patched_motor():
    m = SimMotor("")
    await m.connect(mock=True)
    with patch_motor(m):
        await m.set(10)
        assert await m.user_readback.get_value() == 10
```

```
FAILED tests/core/test_mock_signal_backend.py::test_patched_motor - TypeError: patch_motor.<locals>.<lambda>() got an unexpected keyword argument 'wait'
```

PR #326 (2024-05-22): added a test for mock side effect with misaligned args
Closes #319

I couldn't reproduce the behaviour in 319 for whatever reason. Maybe it's been indirectly fixed since now that we use a MagicMock instead in the mock signal backend.

This PR just adds a test to show that we do get a nice error.
closed 2024-05-22 by evalott100