bluesky/ophyd-async #322: Create helper for mbb(i/o)Direct records

DominicOram, 2024-05-20:
An mbb(i/o)Direct record has the fields B0...BF (see https://epics.anl.gov/EpicsDocumentation/AppDevManuals/RecordRef/Recordref-26.html). It is probably quite likely that people will want to alias these into individual signals. e.g. In the `Attenuator` in `dodal` for the `DEC_TO_BIN` PV we have:
```python
        self.calculated_filter_states: DeviceVector[SignalR[bool]] = DeviceVector(
            {
                int(digit, 16): epics_signal_r(int, f"{prefix}DEC_TO_BIN.B{digit}")
                for digit in string.hexdigits
                if digit.isupper()
            }
        )
```

This is quite a bit of boilerplate that is potentially useful in a generic way e.g. it would be nice if in `dodal` we could do something like:
```python
from ophyd_async.epics import mbbRecord
self.calculated_filter_states: mbbRecord = mbbRecord(f"{prefix}DEC_TO_BIN")
```
to achieve the same result.

coretl, 2024-07-19:
I suggest:
```python
# in ophyd_async.epics.signal
def epics_mbb_record_r(pv: str) -> DeviceVector[SignalR[bool]]:
    return DeviceVector(
            {
                i: epics_signal_r(bool, f"{pv}.B{i:X}")
                for i in range(16)
            }
        )
```

DominicOram, 2024-07-19:
I believe I tried `bool`s, which would be the better solution, and they failed to connect

coretl, 2024-07-19:
> I believe I tried `bool`s, which would be the better solution, and they failed to connect

We should fix that too then...