bluesky/ophyd-async #327: Deal with HDF stall timeout in a better way

coretl, 2024-05-22:
This is the code that applies the timeout:
https://github.com/bluesky/ophyd-async/blob/9f332e389da01b3841c7d000ee7ddfa13c7b3502/src/ophyd_async/core/detector.py#L311-L322

It defaults to DEFAULT_TIMEOUT=10s, which is not enough time for long exposures or gappy hardware triggered scans. Only the trigger source will know a reasonable value for this, so I suggest we don't specify it on detector init, but supply it in `StandardDetector.prepare`, and default it to frame exposure time + DEFAULT_TIMEOUT.

_Originally posted by @coretl in https://github.com/DiamondLightSource/i22-bluesky/issues/13#issuecomment-2125084495_

callumforrester, 2024-05-23:
This does seem like it would take a long time to fail for, say, two clusters of frames separated by a 2 minute gap. I wonder if we should also have the option of passing an array of timeouts, or is that too complex?

danielballan, 2024-05-23:
> Only the trigger source will know a reasonable value for this

I think we came to a similar conclusion about timeouts for `set`. This makes sense and is consistent.

Not sure about array of timeouts...seems complex and potentially mistake-prone in a frustrating way. If the user passes an upper bound, the only downside is sometimes taking too long to give up on real failures, which is should be a rarity?

coretl, 2024-05-23:
I think the majority of use cases will be supported with a single timeout, and those that aren't will be served well enough by a very long timeout, so I think a single timeout is what we should implement

PR #348 (2024-06-04): Add frame_timeout to trigger info
This adds the `frame_timeout` to `TriggerInfo` so that it can be passed in via prepare for flyscans. For stepscans a default set of `TriggerInfo` values are provided.

I can't tell if this has made the tests longer or they already have got long.

Fixes #327 .
closed 2024-06-04 by abbiemery