bluesky/ophyd-async #329: Investigate hanging collect_while_completing planstub

abbiemery, 2024-05-22:
In the `time_resolved_fly_and_collect_with_static_seq_table` plan stub I have had to change the `collect_while_completing` stub call back to the code used for the i22 experiment as the new code just hung.

An issue that was raised back then has not been addressed but i'm not entirely convinced this is the full problem. The issue and following discussion is here https://github.com/bluesky/bluesky/issues/1630.

This needs investigating.

coretl, 2024-07-11:
Reproduced this on P45

ZohebShaikh, 2024-08-22:
It seems to be working with the latest version of ophyd async main `d7f0748 `with the following change

The plan used to test this was 

```
from pathlib import Path
from pprint import pprint

from bluesky.run_engine import RunEngine
from bluesky.plan_stubs import close_run, open_run, prepare, stage_all, unstage_all

import bluesky.preprocessors as bpp
from ophyd_async.core import (
    DeviceCollector,
    StandardFlyer,
    StaticPathProvider,
)
from ophyd_async.core._detector import DetectorTrigger, TriggerInfo
from ophyd_async.core._providers import StaticFilenameProvider
from ophyd_async.epics.adaravis import AravisDetector
from ophyd_async.fastcs.panda import (
    DatasetTable,
    HDFPanda,
    StaticSeqTableTriggerLogic,
)
from ophyd_async.plan_stubs import fly_and_collect
from ophyd_async.plan_stubs._fly import prepare_static_seq_table_flyer_and_detectors_with_same_trigger

RE = RunEngine()
fp = StaticFilenameProvider("ophyd_async_tests")
path_provider = StaticPathProvider(fp, Path("/dls/p45/data/2024/cm37283-2"))

with DeviceCollector():
    panda_1 = HDFPanda(
        "BL45P-MO-PANDA-01:",
        path_provider=path_provider,
    )
    panda_2 = HDFPanda(
        "BL45P-MO-PANDA-02:",
        path_provider=path_provider,
    )
    diff = AravisDetector(
        "BL45P-EA-DIFF-01:",
        path_provider=path_provider,
        drv_suffix="DET:",
        hdf_suffix="HDF5:",
    )

chosen_panda = panda_1
trigger_logic = StaticSeqTableTriggerLogic(chosen_panda.seq[1])
flyer = StandardFlyer(
    trigger_logic,
    configuration_signals=[],
    name="flyer",
)

def plan():
    @bpp.stage_decorator(devices=[diff, chosen_panda, flyer])
    @bpp.run_decorator()
    def inner():
        yield from prepare_static_seq_table_flyer_and_detectors_with_same_trigger(flyer,[diff],number_of_frames=15,exposure=0.1,shutter_time=0.05)
        yield from fly_and_collect(
            stream_name="primary",
            flyer=flyer,
            detectors=[diff],
        )
    yield from inner()

RE(plan(), lambda name, doc: pprint({"name": name, "doc": doc}))

[... 59 lines cut ...]
         'indices': {'start': 4, 'stop': 9},
         'seq_nums': {'start': 5, 'stop': 10},
         'stream_resource': 'b0f85ff3-2d25-4d68-b4d8-e3b3da1cdbca',
         'uid': 'b0f85ff3-2d25-4d68-b4d8-e3b3da1cdbca/1'},
 'name': 'stream_datum'}
{'doc': {'descriptor': 'e9918279-1c9d-4b81-989a-43a8ddb0f739',
         'indices': {'start': 9, 'stop': 14},
         'seq_nums': {'start': 10, 'stop': 15},
         'stream_resource': 'b0f85ff3-2d25-4d68-b4d8-e3b3da1cdbca',
         'uid': 'b0f85ff3-2d25-4d68-b4d8-e3b3da1cdbca/2'},
 'name': 'stream_datum'}
{'doc': {'descriptor': 'e9918279-1c9d-4b81-989a-43a8ddb0f739',
         'indices': {'start': 14, 'stop': 15},
         'seq_nums': {'start': 15, 'stop': 16},
         'stream_resource': 'b0f85ff3-2d25-4d68-b4d8-e3b3da1cdbca',
         'uid': 'b0f85ff3-2d25-4d68-b4d8-e3b3da1cdbca/3'},
 'name': 'stream_datum'}
{'doc': {'exit_status': 'success',
         'num_events': {'primary': 15},
         'reason': '',
         'run_start': '2fe63d78-3119-47c7-bd03-aab150c745d4',
         'time': 1724338722.5455422,
         'uid': '31366982-b266-4953-81e4-fbb7bf14a7f1'},
 'name': 'stop'}
```

stan-dot, 2024-11-18:
@abbiemery does it mean that it's no longer a problem?