bluesky/ophyd-async #336: Add a shared parent mock for all signals in once device

DominicOram, 2024-05-24:
As a developer I occasionally want to make sure that one signal is changed before another. For example, see `test_retrieve_mock_and_assert` in https://github.com/bluesky/ophyd-async/pull/335/files or for a more real world example see numerous tests in https://github.com/DiamondLightSource/dodal/blob/main/tests/devices/unit_tests/test_aperture_scatterguard.py. To do this you need to create a parent mock that all the signals are children of.

If `ophyd_async` by default made all the mock's in one device have the same parent this would remove a bit of boilerplate from downstream.

coretl, 2024-06-24:
A suggestion for implementation:
- Change `Device.connect` to take `mock: Union[bool, Mock] = False`
- If `isinstance(mock, Mock)` then use it as the mock for the `Device`, otherwise create a new one
- When calling `child.connect`, then call `mock.attach_mock(child_name, child_mock)` and pass `mock=child_mock` to the child

This will mean that if you have a tree of devices where connect it called at the top level then you can call:
```python
get_mock_put(mydevice).assert_has_calls(
        [
            call.child1.grandchild1(100, wait=True, timeout=ANY),
            call.child2.grandchild2(67, wait=True, timeout=ANY),
        ]
    )
```
I'm a little concerned that we might decide we want to check the ordering of puts and other methods like `get_descriptor`, so maybe we should consider this before making a mock heirarchy we rely upon?

For instance, would we ever want something like this?
```python
get_mock(mydevice).assert_has_calls(
        [
            call.child1.grandchild1.put(100, wait=True, timeout=ANY),
            call.child1.grandchild1.get_descriptor(),
            call.child2.grandchild2.put(67, wait=True, timeout=ANY),
        ]
    )
```

@DominicOram thoughts?

coretl, 2024-07-19:
@DominicOram any thoughts on this?

coretl, 2024-07-22:
@dperl-dls any thoughts on this?

DominicOram, 2024-07-22:
>A suggestion for implementation:

Yes, I like this idea.

> I'm a little concerned that we might

This feels like feature creep at this point. I can't see a reason we would want to do this yet and it doesn't seem like the way we structure this hierarchy will have that much of an affect on how we want to do it in the future.

coretl, 2024-07-22:
> This feels like feature creep at this point. I can't see a reason we would want to do this yet and it doesn't seem like the way we structure this hierarchy will have that much of an affect on how we want to do it in the future.

Ok, let's go with the minimum change then, so

```python
get_mock_put(motor.setpoint).assert_called_once_with(32.0, wait=True, timeout=ANY)
```

will still work, and so will:

```python
get_mock_put(motor).assert_has_calls(
        [
            call.setpoint(32.0, wait=True, timeout=ANY),
            call.velocity(1.0, wait=True, timeout=ANY),
        ]
    )
```

jsouter, 2024-10-10:
Struggling to get a test to work here, it seems like copying this test https://github.com/bluesky/bluesky/blob/b6828c53a68252caeefc3712ea50c49e507d8db2/src/bluesky/tests/test_suspenders.py#L37 

and replacing the `threading.Timer`s with `asyncio.create_task`s with the coroutine

```
    async def _set_after_time(val, sleep):
        await asyncio.sleep(sleep)
        await signal.set(val)
```
and starting those tasks before calling 
```
    scan = [Msg("checkpoint"), Msg("sleep", None, 0.2)]
    RE(scan) 
```
that the RE finishes its scan before control returns to the asyncio tasks. I've tried in a few different ways but maybe there's a much more obvious way to test this. The suspender's resume/suspend methods get called at least but I haven't been able to prove it interrupts as intended. 

I have also seen that when trying to set a signal to a value that should suspend while inside a plan using 
`yield from bps.abs_set(signal, fail_val)` that `bluesky.suspender.SuspenderBase.__make_event` fails to create an `asyncio.Event` -- `h = self.RE._loop.call_soon_threadsafe(really_make_the_event)` seems to never actually call the function so h gets cancelled.

coretl, 2024-10-14:
> Ok, let's go with the minimum change then, so
>
> ```python
> get_mock_put(motor.setpoint).assert_called_once_with(32.0, wait=True, timeout=ANY)
> ```
>
> will still work, and so will:
>
> ```python [...]

I'm going to roll back on this, as I figure we can support a future addition of `mock.get_descriptor` with minimal effort.

If we make it so:
- `Device.connect` is either passed a `Mock`, or creates it if `mock is True`
- Same for `Signal.connect`
- `MockSignalBackend.put` calls `mock.put()`
- In the future `MockSignalBackend.get_descriptor()` could call `mock.get_descriptor()`
- `get_mock_put` returns `backend.mock.put`
- We add a new `get_mock` that returns `backend.mock`
- Existing tests do not need to change
- New tests can do:

```python
get_mock(mydevice).assert_has_calls(
        [
            call.child1.grandchild1.put(100, wait=True, timeout=ANY),
            call.child2.grandchild2.put(67, wait=True, timeout=ANY),
        ]
    )
```

Do we lose anything with this approach?

jsouter, 2024-10-17:
@coretl Working on this, do we want to pass the same Mock from the Device -> Signal -> MockSignalBackend (either by adding a new arg to the init or with a set_mock call maybe) or should the MockSignalBackend's `self.mock` be always set to a new
`self.mock = Mock()` before calling `self.mock.put = AsyncMock(name="put", spec=Callable)? I'm not sure if we'd want all the backends to have the same put mock or not.

jsouter, 2024-10-18:
I believe this is working as intended now, I will attempt to resolve the conflicts

coretl, 2024-10-18:
> I believe this is working as intended now, I will attempt to resolve the conflicts

Unfortunately I've changed the connect structure (again), sorry!

coretl, 2024-10-18:
now is good, I don't think I need to change the connect logic again

closed 2024-10-31 by coretl