bluesky/ophyd-async #351: `set_mock_values` does not work with `observe_value`

callumforrester, 2024-05-31:
As a developer I would like to inject sequences of values into mock signals that can be read out by a monitor so I can simulate streamed behaviour.

Currently if I call `set_mock_values` on a signal and observe it, it yields nothing and the mock values are never consumed. I am unsure if this is expected behaviour.

Here is a failing test:
```python
async def test_set_mock_values_with_observe(mock_signals):
    signal1, _ = mock_signals
    set_mock_values(
        signal1,
        ["first_value", "second_value", "third_value"],
        require_all_consumed=True,
    )
    values = []
    async for value in observe_value(signal1, timeout=1.0):
        values.append(value)
    assert values == ["first_value", "second_value", "third_value"]

```

## Acceptance Criteria
- The above test passes

coretl, 2024-06-03:
@evalott100 added some expected usage in the docstring, but the docs could probably be improved:
https://github.com/bluesky/ophyd-async/blob/b2b8bba34eed7bd8d7ee55a594a4fcfe5f4be888/src/ophyd_async/core/mock_signal_utils.py#L121-L124

So we use an `async for` to trigger the next mock value. E.g.
```python
async def my_func(sig, values):
    async for value in observe_value(sig):
        values.append(value)
        if value == "2nd":
            break

values = []
task = asyncio.create_task(my_func(signal1, values))
async for expected_value in set_mock_values(signal1, ["1st", "2nd"]):
    assert values[-1] == expected_value
# Might need an asyncio.sleep(A_BIT) here...
assert task.done
```
or if you aren't testing an external function and can interleave test code:
```python
values = []
signal1.subscribe_value(values.append)
async for expected_value in set_mock_values(signal1, ["1st", "2nd"]):
    assert values[-1] == expected_value
signal1.clear_sub(values.append)
```

callumforrester, 2024-06-03:
@coretl How would that work if I were testing with the `RunEngine`? Say I have a device that waits for a set of values internally (e.g. armed then disarmed) when triggered? I would use the `RunEngine` synchronously so couldn't just kick off a task in the background.

coretl, 2024-06-05:
Is there a test plan, or a production plan? If there is a test plan then it will run under the RE event loop, so can spawn the task from there. If not, then either wrap the production plan in a test plan where you spawn, or do `call_in_bluesky_event_loop` with a coro that makes a task

callumforrester, 2024-06-05:
Yes, it's a test plan. Not sure I understand, though. If it gives more context I was hoping to do something like

```python
def test_count(RE):
    with DeviceCollector(mock=True):
        det = MyDetector(...)
    set_mock_values(det.drv.acquire, [True, False])
    RE(count([det]))
```

Do we have a general approach for this?

coretl, 2024-06-06:
Current approach is to hook these things as a callback to something that has been run in the plan:

```python
def test_count(RE):
    with DeviceCollector(mock=True):
        det = MyDetector(...)
    callback_on_mock_put(det.drv.acquire, lambda v, **_: set_mock_value(det.drv.acquiring, v))
    RE(count([det]))
```

like in:
https://github.com/bluesky/ophyd-async/blob/0b05b9c0c7c25417cb76432de0eae067fc7a86ce/tests/panda/test_hdf_panda.py#L42-L45

@abbiemery when you go through tidying tests it might be worth a review of the test functions in https://github.com/bluesky/ophyd-async/blob/main/src/ophyd_async/core/mock_signal_utils.py to check that they are as useful as they could be.

closed 2024-06-06 by coretl