bluesky/ophyd-async #381: Provide a way to disarm `StandardDetector`s from plans

callumforrester, 2024-06-12:
As a user I would like to write plans that attempt to use detectors but can disarm them in the event of an error so that I don't leave my beamline in an expected state. For example

```python
def my_plan(detectors):
    yield from scan(*detectors, ...)
```

The above plan would unstage the detectors if there was an error but `StandardDetector`s only disarm their file writing when unstaged. The expect the IOC to disarm the detector once acquisition is complete, but if an error occurs, it never completes.
There are good reasons for this behaviour; leaving the beamline in a broken state my sometimes be desirable for diagnostic reasons, it may also be desirable to try again several times and leave the detector armed for performance reasons (@DominicOram is likely to have this use case).

Perhaps the solution looks something like this?

```python
def my_plan(detectors):
    def inner():
        yield from scan(*detectors, ...)
    def cleanup():
        yield from disarm(*detectors)
```

Would be interested in @coretl's thoughts.
This may also relate to #314 

## Acceptance Criteria
- There is a way to write plans that will disarm their detectors if something goes wrong
- This behaviour is not enforced by default
- There are tests to cover it

DominicOram, 2024-06-12:
I think we would actually like to always disarm on error. We're picky on disarming in general and may want to chain collections w/o disarming but if one of them genuinely fails then I think it's fair to completely stop and get us into a cleaner state.

coretl, 2024-06-14:
I have the opposite requirement, when something fails I want a mode to leave it in that state so I can poke EPICS further! It was immensely frustrating when GDA ran cleanup code on error as we could never prod the system when it was in an error state.

Malcolm's behaviour is to fail and leave everything in that state, returning to the user, then put things back into a known good state when you call `reset()`. Is there anything wrong with that behaviour?

If we prefer to disarm on error, then I'd like a way to turn that off...

coretl, 2024-06-17:
New suggestion: could have a mode on the RE that converts errors into a Pause method, that would allow us to see what happened

callumforrester, 2024-06-18:
So the specific case I had in mind was that leaving pilatuses armed for too long without taking frames can cause problems with the camserver. Depending on the firmware it can take about 10-20 minutes. It's a bit useless to leave them armed for poking if the controls engineer takes longer than that to arrive. 

It might actually be a special case though, I generally agree with @coretl's assertion that we shouldn't touch much in the event of an error, so maybe we need to either:
- Have a special way to disarm only pilatuses in the event of an error
- Fix the epics/camserver interaction so this isn't a problem in the first place

coretl, 2024-06-18:
How about this:
- We `disarm` in unstage
- @DominicOram can you do multiple collections without calling `unstage`?
- The top level plan will call `unstage` automatically on error as they are using the stage decorator
- We have our new mode on the RE that puts it into Pause on error
- We only enable that mode when we are actively down on the beamline debugging things

DominicOram, 2024-06-18:
> @DominicOram can you do multiple collections without calling `unstage`?

Yes, that was going to be the plan.

I think I agree with everything discussed on this issue. A pause on error mode would be useful