bluesky/ophyd-async #382: PandA HDFWriter stream_resource path is incorrect for PCAP block

DiamondJoseph, 2024-06-12:
For a received stream_resource

```json
{
    "name": "stream_resource",
    "doc": {
        "uid": "a62f9676-0652-41be-a87a-2918295ee027",
        "data_key": "panda1-pcap-samples-Value",
        "spec": "AD_HDF5_SWMR_SLICE",
        "root": "/scratch/docs/12-6",
        "resource_path": "/scratch/docs/12-6/i22-3-panda1.h5",
        "resource_kwargs": {
            "name": "panda1-pcap-samples-Value",
            "block": "pcap",
            "path": "PCAP-SAMPLES-Value",
            "multiplier": 1,
            "timestamps": "/entry/instrument/NDAttributes/NDArrayTimeStamp"
        },
        "path_semantics": "posix",
        "run_start": "0362ddf7-8f09-4539-9288-eb303e66c059"
    }
}
```

the produced h5 file had the follwoing structure
[image]

coretl, 2024-06-18:
Is this fixed now?

DiamondJoseph, 2024-07-02:
This is now fixed

closed 2024-07-02 by DiamondJoseph