bluesky/ophyd-async #387: Numpy 2 compatibility

callumforrester, 2024-06-17:
The current main error that numpy 2 seems to cause:

```shell
Traceback (most recent call last):
  File "/home/vid18871/projects/bluesky/ophyd-async/venv/lib64/python3.10/site-packages/_pytest/runner.py", line 341, in from_call
    result: Optional[TResult] = func()
  File "/home/vid18871/projects/bluesky/ophyd-async/venv/lib64/python3.10/site-packages/_pytest/runner.py", line 389, in collect
    return list(collector.collect())
  File "/home/vid18871/projects/bluesky/ophyd-async/venv/lib64/python3.10/site-packages/pytest_asyncio/plugin.py", line 640, in _patched_collect
    module = collector.obj
  File "/home/vid18871/projects/bluesky/ophyd-async/venv/lib64/python3.10/site-packages/_pytest/python.py", line 287, in obj
    self._obj = obj = self._getobj()
  File "/home/vid18871/projects/bluesky/ophyd-async/venv/lib64/python3.10/site-packages/_pytest/python.py", line 545, in _getobj
    return importtestmodule(self.path, self.config)
  File "/home/vid18871/projects/bluesky/ophyd-async/venv/lib64/python3.10/site-packages/_pytest/python.py", line 492, in importtestmodule
    mod = import_path(
  File "/home/vid18871/projects/bluesky/ophyd-async/venv/lib64/python3.10/site-packages/_pytest/pathlib.py", line 591, in import_path
    importlib.import_module(module_name)
  File "/usr/lib64/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "/home/vid18871/projects/bluesky/ophyd-async/venv/lib64/python3.10/site-packages/_pytest/assertion/rewrite.py", line 178, in exec_module
    exec(co, module.__dict__)
  File "/home/vid18871/projects/bluesky/ophyd-async/tests/sim/test_sim_detector.py", line 4, in <module>
    from ophyd_async.epics.motion import motor
  File "/home/vid18871/projects/bluesky/ophyd-async/src/ophyd_async/epics/motion/__init__.py", line 1, in <module>
    from .motor import Motor
  File "/home/vid18871/projects/bluesky/ophyd-async/src/ophyd_async/epics/motion/motor.py", line 19, in <module>
    from ..signal.signal import epics_signal_r, epics_signal_rw, epics_signal_x
  File "/home/vid18871/projects/bluesky/ophyd-async/src/ophyd_async/epics/signal/__init__.py", line 1, in <module>
    from .signal import (
  File "/home/vid18871/projects/bluesky/ophyd-async/src/ophyd_async/epics/signal/signal.py", line 17, in <module>
    from ._epics_transport import EpicsTransport
  File "/home/vid18871/projects/bluesky/ophyd-async/src/ophyd_async/epics/signal/_epics_transport.py", line 17, in <module>
    from .._backend._p4p import PvaSignalBackend
  File "/home/vid18871/projects/bluesky/ophyd-async/src/ophyd_async/epics/_backend/_p4p.py", line 11, in <module>
    from p4p import Value
  File "/home/vid18871/projects/bluesky/ophyd-async/venv/lib64/python3.10/site-packages/p4p/__init__.py", line 14, in <module>
    from .wrapper import Value, Type
  File "/home/vid18871/projects/bluesky/ophyd-async/venv/lib64/python3.10/site-packages/p4p/wrapper.py", line 5, in <module>
    from . import _p4p
  File "src/p4p/_p4p.pyx", line 1, in init p4p._p4p
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject

```

callumforrester, 2024-06-17:
Solution here is likely to wait for p4p (and possibly other upstream libraries) to be recompiled against numpy 2.0

DominicOram, 2024-06-17:
See https://github.com/mdavidsaver/p4p/issues/145

closed 2024-11-22 by coretl