bluesky/ophyd-async #393: Cannot have tuples as soft signal values

callumforrester, 2024-06-17:
Trying to make a soft signal of type `tuple[int, str, float]`, say, yields an error when you try to `count` it:

```shell
AssertionError: invalid converter for value of type <class 'tuple'>
```

coretl, 2024-06-18:
I think this is a feature not a bug. Please can you provide the use case for supporting it.

callumforrester, 2024-06-18:
I have a soft signal whose value should be a 3d vector (miller indices). So I wanted a `SignalR[tuple[int, int, int]]`. I can have a list, but then the type checker won't validate the dimensionality. Is there a particular issue with supporting tuples?

coretl, 2024-06-18:
We had an ambition to do the type checking at the Signal level rather than the SignalBackend level. Then we didn't need to define another `TypeVar` for the supported types of each SignalBackend. Thinking about it though, PVA is more restrictive than CA so we would have to do that anyway.

Ok, so this makes sense. I assume the usecase is a SoftSignalBackend. How much runtime error checking do we want for those? Or casting? Length checking?

callumforrester, 2024-06-19:
...I'm happy with virtually none. Unlike with an epics backend I'm typing the values into my IDE and using a static type checker so I'm happy to rely on that.

coretl, 2024-06-19:
I guess your use case is read only then? For the read-write case, then your user might be using a generic `abs_set` plan, so will be reliant on runtime checking...

callumforrester, 2024-06-20:
I will also say my use case for code that is temporary and that we don't expect to last forever, so if no one else is asking for this I wouldn't prioritise it.